fairscope/PlanktoScope #8: Strobing the LED?

rljacobson, 2020-08-16:
Have you experimented with strobing the LED to increase image clarity and/or capture rate?

tpollina, 2020-09-17:
We discussed that possibility internally in the PrakashLab a year ago as a possibility to increase the capture rate which is quite low with the current code. But since we considered a deployment on sailboat as one of the main application with a low constraint on acquisition time, we haven't decided to prioritize the capture rate yet.

But in the scenario that you were imagining, if the particles are moving (even if we stop the flow, the gravity is still acting + living cells can move around) and the camera sensor is a global shutter (PiCamera V2.1), how can we increase image clarity / capture rate with strobing the LED ?

Could be amazing to dig into that direction especially if people like you could suggest a way to program this ?

rljacobson, 2020-09-18:
Image clarity is presumably limited by motion blur. A faster shutter speed will reduce motion blur at any frame rate. A strobe has the effect of increasing the shutter speed by illuminating the scene only for the duration of the "effective" shutter speed.

Both Raspberry Pi cameras have a rolling shutter. The v.1 camera has a _global reset_, though, which means the pixels are exposed simultaneously. The flash acts like a shutter. 

The v.2 camera does not have a global reset. But it should still be fast enough, as it can do ~1000fps under the right conditions. ([See here](https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=234416&p=1434419#p1433947).) I'm not sure I know enough about the v.2 camera's theory of operation to say how a strobe will be effective, but others on the net have reported a reduction of motion blur with a strobe.

There are genuine global shutter cameras for the Raspberry Pi from [ArduCam](https://www.arducam.com/products/camera-breakout-board/global-shutter-camera/). These are unlikely to be the right thing for this application.

fabienlombard, 2020-09-20:
Well global shutter or flashing, we need to capture at high speed to get nicely the swimming organisms (plus increasing the speed of acquisition)

moi90, 2020-09-25:
It could be very hard to synchronize the LED flash with the camera shutter. At my lab, we have a special camera that produces its own flash signal (which can be controlled in length and positive/negative delay). Doing this in software will most likely not be accurate enough.

Also, if the pixels are not exposed simultaneously, a flash will not help much, I suspect. The question is: Do the exposure time slots of the individual rows overlap enough so that a flash can be short but cover all of them simultaneously?

I think, a global shutter camera would be the most robust option.

@rljacobson Why do you think that the global shutter cameras from ArduCam won't work?

moi90, 2020-09-25:
Apparently, there is a built in hardware flash signal: https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=83484

gromain, 2020-09-25:
Yes, there is. It needs a specially compiled device tree file though. The picamera documentation has some information on that too: https://picamera.readthedocs.io/en/release-1.13/recipes2.html?#using-a-flash-with-the-camera

gromain, 2020-09-25:
Also, yes, line by line, the data is coherent. Once again, the picamera documentation has all the good stuff: https://picamera.readthedocs.io/en/release-1.13/fov.html

rljacobson, 2020-09-25:
@moi90 
> Why do you think that the global shutter cameras from ArduCam won't work?

Because the benefit of the global shutter is probably outweighed by their inferior resolution. I think they would work. I just expect they would be much worse than what you already have. It's a guess, of course. I haven't tried it. 

I would be shocked if you couldn't get the timing down to well within the tolerances required here even without hardware support. We're talking milliseconds, not nanoseconds. A millisecond is a lifetime for a microcontroller. But again, I haven't tried it. Others on the internet seem to have had success with strobing in other contexts. A concern I have is whether or not the generic LEDs you can get in an off-the-shelf electronics kit can strobe fast enough. Maybe, maybe not. If not, that's another specialty component required.

gromain, 2020-09-27:
I don't think the LEDs will have a problem with being strobed really fast. I used to drive some at 30kHz without issue, and they were just standard LEDs.
I believe that if we want to experiment with this, the flash pin method is the right route to follow and would allow us to achieve what we want.
Right now, with the LED I sourced, I can go for a shutter speed as low as 50µs with 100 ISO (so probably much lower if I augment the ISO level reasonably so we don't have much noise).

moi90, 2020-09-29:
All in all, it sounds that a flash can and should be used, right?

gromain, 2020-09-29:
At least, I believe it's something that should be tested!

gromain, 2020-09-29:
So, I quickly created a new dt-blob binary for the PlanktoScope to test the LED flash. It seems to be working great. I just have to test the capture on a sample, without the flash, and with the flash, to see if it's worth using it.
I don't see a way of controlling the flash length though, so there's that.

fabienlombard, 2020-09-29:
> Links:

Do you have a working code, and a easy way to implement it? I still have 
the Lugol fixed sample so a test is easy to operate

Fabien

moi90, 2020-10-02:
If I understand the documentation correctly, the flash duration is the (manually or automatically) configured shutter_speed. So I would not expect an improvement when using the flash compared to constant illumination.

The only advantages of strobing would be: a) Because it is only switched on for a short time, the LED can be powered with more current than it is specified for, so more photons per time. b) The LED draws less power overall, which could be advantageous when running on battery. c) The LED does not need a heat sink.

How fast is the exposure_speed currently? Is it set automatically or configured to a fixed value?

I think the best way to proceed would be to set the shutter_speed to the absolute minimal value and use a really high power LED that floods the scene with photons. Remaining rolling-shutter artifacts then have to be corrected in software (which is possible).

fabienlombard, 2020-10-02:
Well

The camera is currently a rolling shutter camera (generating some weird 
deformation when animals move , which happens quite a lot when the fluid 
is pumped, forcing us to use a stop and go method), so flash could give 
the possibility to obtain something coherent with a global shutter 
camera without upgrading the camera itself, and would allow for a 
quicker acquisition without the stop and go

moi90, 2020-10-03:
Yes and a flash won't magically transform a rolling shutter camera into a global shutter camera. If the camera had a *global reset shutter*, it could work (like shown [here](https://www.baumer.com/de/en/service-support/know-how/function-principle/function-principle-and-applications-of-rolling-shutter-cmos-cameras/a/CMOS-rolling-shutter-cameras)).

As @rljacobson said:

> Both Raspberry Pi cameras have a rolling shutter. The v.1 camera has a global reset, though, which means the pixels are exposed simultaneously. The flash acts like a shutter.
>
> The v.2 camera does not have a global reset.

Let's see how the tests turn out.

(As I said: My guess is that a high-power LED allows us to reduce the exposure time to a minimum (and therefore minimize the delay between the exposure of subsequent lines and also the influence of stray light) and that we can use post-processing to correct for the remaining motion artifacts.)

gromain, 2020-10-03:
From my tests, I do not think flashing will improve anything (because of the global reset thingy mainly, but also because the flash duration seems to be like the shutter duration).
I'm able to image at 10µs with ISO 500. I can probably go lower with a higher output LED (that probably can't be drived directly by the Raspberry).
My LED output is at 40000 mcd with a 15° angle. I can try to source something more powerful with the same angle (or with a lens to achieve the same angle).

moi90, 2020-10-05:
Yes, even if the RPi would be able to provide enough current, it would be better to use an adjustable constant current driver (like [this one](https://www.onsemi.com/products/power-management/led-drivers/linear-led-drivers/nsi50150ad?utm_medium=ppc&utm_source=google&utm_campaign=dsa%20germany%20conversions&utm_term=&utm_keyword=&gclid=CjwKCAjwiOv7BRBREiwAXHbv3L72XDrxLYS2-la8N8RApcs4LzsvmuPq6shQFaGUfemPiEMe0lDpdxoCLOoQAvD_BwE)).

gromain, 2020-10-05:
I agree. I have in mind to have a dedicated line on a custom made HAT that PlanktonPlanet could provide to help lower build costs.
Definitely something I'm looking at seriously!

gromain, 2021-12-01:
For the record, the custom hat has a way to strobe the LED. However, minimal pulse length is 40ms, which may be way too long.

We should investigate with fast switching of the LED selection output.