fairscope/PlanktoScope #10: Test Kamoer pump

gromain, 2020-09-17:
We need to source an appropriate peristaltic pump, with consistent quality and availability.
Kamoer pumps seems to be appropriate candidates (see  the catalog at http://kamoer.com/upload/file/201910/1571712939838489.pdf ).

We need to choose one or two models, get them to run 24/7 and see their lifetime (in terms of total volume pumped or something similar).

gromain, 2020-10-21:
First tests of this pump are quite conclusive. However, the model I choose initially is quite large. I'm ordering now a smaller one but I'm confident this would be a good fit for the machine.

gromain, 2020-11-24:
Apart from the volume pumping test, this is mostly done!

ethanjli, 2023-10-26:
The newer versions of the PlanktoScope hardware (including v2.5 and v2.6) use the Kamoer pumps. This issue appears to have been resolved already, so I will close it.

closed 2023-10-26 by ethanjli