fairscope/PlanktoScope #11: Alternative hat for stepper control

gromain, 2020-09-17:
We need to improve steps precision and speed, eventually acceleration/deceleration for the pump.
Also, we need to have control over the maximum current draw (today, there is no limit and it happened that a stepper hat was fried due to an overcurrent)

https://www.waveshare.com/stepper-motor-hat.htm
https://www.st.com/en/evaluation-tools/evalsp820-xs.html#

gromain, 2020-11-18:
The waveshare stepper hat is good, however, there is one problem:
The inputs to the controller are left floating by the raspberry by default. This means they enter an enabled state and the steppers are powered and heating. This can lead to a destruction of the steppers. This has been addressed partly by adding an initialisation in `/boot/config.txt` but as soon as the main user is logged in, the steppers are enabled again (so in Node-Red, we need to have something to also initialise those outputs to disable the steppers).

All in all, a good candidate, but this problem is not nice.
Another controlled is going to be tested, consisting of TMC5160 controlled by the SPI bus.

ethanjli, 2023-09-18:
I am closing this issue because it appears to be done: recent versions of the PlanktoScope hardware (since hardware v2.3) now use a custom HAT for stepper control.

closed 2023-09-18 by ethanjli