fairscope/PlanktoScope #12: Draw logic block diagram of hardware components

gromain, 2020-09-17:


ethanjli, 2023-11-10:
Architecture diagrams will be needed for a "Technical Reference" section of our new documentation site. We will probably need block diagrams at various levels of detail. I suggest starting with [functional block diagrams](https://en.wikipedia.org/wiki/Functional_block_diagram) showing the subsystems of the PlanktoScope and their interactions. It would be ideal if we could generate diagrams from plain text, so that they can be version-controlled in Git.

chevreuill3000, 2026-11-09:
Closing as not planned: there has been no activity on this issue for a long time and it is not on the current roadmap. Feel free to reopen it if it is still relevant.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-11-09 by chevreuill3000