fairscope/PlanktoScope #14: Waterproofing electronics

gromain, 2020-09-17:
We need to find a way to waterproof the electronics. Doesn't need to be IP69, but at least resistant to spray.

A good way to do this would be to use a waterproofing spray (some kind of water repellent spray). A simple very low cost and widely available solution could be to use a fine mist of WD40 (this doesn't dry, so it will leave the electronics kind of moist to the touch).

gromain, 2020-09-17:
Products to test:
- https://blog.adafruit.com/2013/09/04/nanoprotech-waterproof-coating-wearablewednesday/
- https://www.amazon.fr/MG-Chemicals-Tropicalisation-Silicone-Modifi%C3%A9/dp/B008O9YGQI see here https://www.youtube.com/watch?v=HOu5hRqSDtU (en pot: https://www.amazon.fr/Rev%C3%AAtement-conforme-422C-indicateur-Bouteille/dp/B089XXS72K )
- https://www.amazon.fr/Isolant-électrique-Plasti-Dip-transparent/dp/B01G4ABSVY/
- https://www.amazon.fr/acrylique-isolant-Tropicoat-Jelt-007361/dp/B008BT4YS6/
- https://www.amazon.fr/PROTECTION-SPECIAL-CIRCUIT-ELECTRONIQUE-MECANIQUE/dp/B07BR61KPB

fabienlombard, 2020-09-20:
The metallic plate supporting the focus motors seem to be an obvious weak point for this

ethanjli, 2023-11-10:
As far as I'm aware, this is not part of the scope of hardware development which @tpollina is pursuing. I expect that the question of making the PlanktoScope robust in environments which subject it to water splashes/sprays/immersion will eventually become relevant again in the future, but for now I will close this issue as wontfix; once multiple people request some sort of mitigation for water exposure, we can either reopen this issue or open a new issue.

closed 2023-11-10 by ethanjli