fairscope/PlanktoScope #16: Store metadata information in EXIF tags

gromain, 2020-09-17:
Allows for a better traceability of metadata solely based on the image file.

Tag details are at https://www.exiv2.org/tags.html

moi90, 2020-09-25:
The usage of EXIF (or XMP or something else) would indeed make a index.csv (or the like) in the output archive obsolete and keep the information tightly attached to an image file.

gromain, 2020-09-25:
We thought about it and we think it makes sense mainly if the archive uses this data. Otherwise, this may be a bit useless.

moi90, 2020-09-27:
So you mean regular users would be better off with a plain csv file for metadata? I agree, as long as the csv file is not separated from the images.

I doubt that we should store images individually in an archive but rather as a zip file (or similar). So an archive would not read the data anyways.

Should we close this then?

gromain, 2020-09-27:
Good for me. I'll close it now. Eventually, we'll reopen if at some point it makes sense to have that (maybe for the GPS data at least?)

closed 2020-09-27 by gromain