fairscope/PlanktoScope #19: Write a detailed getting started guide

gromain, 2020-09-17:
After assembly and setup of the Planktoscope, it is not clear the steps that should be followed.

We need to write a guide to cover those steps:
- full tests
- first pictures
- calibration of the machine
- plankton collection with nets
- basic and advanced workflow for data and image creation

ethanjli, 2024-04-04:
As discussed in the [2024-03-28 software meeting](https://docs.google.com/document/d/1Iba0_9yNFEDdp1Qy8IYlLP43Ga0DnDr3CdavWV0QGZY/edit#heading=h.2qje39kpb6c2), for now @pkostyrka and @fabienlombard's [protocols on protocols.io](https://www.protocols.io/view/planktoscope-protocol-for-plankton-imaging-bp2l6bq3zgqe/) are serving the purpose of a detailed "getting started" guide to help users acquire their first image dataset; it'd be nice to also include a complete and up-to-date getting-started guide on https://docs.planktoscope.community/ , but it's a lower priority because for now we can just link users to the protocols.io protocols.

chevreuill3000, 2026-11-09:
Closing as not planned: there has been no activity on this issue for a long time and it is not on the current roadmap. Feel free to reopen it if it is still relevant.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-11-09 by chevreuill3000