fairscope/PlanktoScope #21: Replacement for ibidi flow cell

gromain, 2020-09-17:
Ibidi flow cells are expensive, hard to find and quite fragile.

A proposed inexpensive alternative is to use a flat capillary glass tube. Suppliers are still being searched.
A 3D printed support for this tube could be used to mount it in place of the ibidi flow cell.

fabienlombard, 2020-09-20:
Noan le Bescot have a company able to do this plus a way to glue the capillary (although needing long 3weeks drying time)

gromain, 2020-09-21:
I'll ask him the information about the supplier then.
This definitely can be linked to #31 and for me goes hand in hand (since we need a way to mount the capillary tube anyway).
We'll also have to design something to limit the stage movement so the flow cell cannot crash against the lens.

ethanjli, 2023-09-18:
I am closing this issue because Thibaut marked it as done, and this improvement clearly has already been implemented in the recent versions of the PlanktoScope hardware.

closed 2023-09-18 by ethanjli