fairscope/PlanktoScope #26: Unique identifier for the export zipfile

tpollina, 2020-09-18:
**Describe the bug**
The archive created by **MorphoCut** has a fix name `export_ecotaxa.zip` and is being overwritten during each segmentation.

**Expected behavior**
Rename `export_ecotaxa.zip` by a unique identifier including `project_name`, `datetime` and `sample_id`.

gromain, 2020-09-20:
This is being addressed in PR #7 , see commit 27d48e1, line [190](https://github.com/PlanktonPlanet/PlanktonScope/commit/27d48e102550b51baca79c2f3cf86d51678e48de#diff-31b22dd05e25f5c75d5a6237b2f9e2e6R190).

moi90, 2020-09-25:
I think, every unit should have a unique ID (e.g. WIFI MAC) and a non-resettable serial number generator so that each sample can be uniquely identified. User-provided project_name, datetime and sample_id are not reliable and should only be used in addition.

gromain, 2020-09-25:
Do you think the unique ID should appear in the filename or in the metadata?

gromain, 2020-09-25:
I'm thinking that maybe a clean sha1 hash of the uniqueID and datetime of the sample should be enough for each sample ID no?

closed 2020-09-28 by gromain