fairscope/PlanktoScope #30: stepper motor are prone to stop

fabienlombard, 2020-09-20:
The bottom stepper motor for focus can block (dust on it?). A gentle manual help unblock it

We should get a protection to this to avoid mist and dust on the focus structure

gromain, 2020-09-21:
Maybe you can try a drop of white lithium grease or silicon grease. This should be harmless to the plastic mounts yet allows for proper lubrication of the threads.

Avoid WD40 as it will attack the plastic!

gromain, 2020-09-21:
Also, an very different alternative solution for this would be to get rid of the stepper control of the focus stage entirely.
This is linked to issue #39. The proposed solution there could simplify mechanical design and reliability.

ethanjli, 2023-11-10:
Given the hardware changes to the focusing stepper motors which have been made in the past 3 years (esp. with the different stepper motors in v2.5 and v2.6 of the hardware), I'm not sure this issue is still relevant. Feel free to make a new issue or reopen this issue with more details on the specific problem, if it's still relevant to v2.5+ of the hardware.

closed 2023-11-10 by ethanjli