fairscope/PlanktoScope #32: Reset to recommended settings button

fabienlombard, 2020-09-20:
It is easy to play with settings, but we should add way to return to « recommended » settings

Bonus: if the settings could be saved in an easy configuration files, users could save their preferred settings (and we could provide several preferred depending on the lens)

gromain, 2020-09-20:
This is definitely something I have been missing since I've started playing with the planktoscope! Thanks for putting it in words!

Do you think this would be needed before the Tara onboard test?

I don't see it as something difficult to manage though, I'd put it at about a day of work and a day of testing.

fabienlombard, 2020-09-21:
> Links:

This is the advantage to have play with many different instruments, 
there is some features that save your life

In term of timing, well, I think that persons onboard tara can follow a 
protocol, and if this is not integrated, we should at least provide a 
fact sheet with recommended settings (especially if we provide access to 
the hard coded settings)

Fabien

gromain, 2020-09-21:
Agreed. For Tara, we can also have a way to hard reset the settings to know good values. This is not very difficult to do. Could be as simple as `git reset --hard` in the PlanktoScope directory.

ethanjli, 2023-11-22:
Since there has been no follow-up on this issue in the past two months indicating the need for any further action beyond what has already been implemented, I will close this issue as completed. If functionality to save custom settings to the filesystem and load them back later is desirable, whoever wants that functionality should create a new feature request.

closed 2023-11-22 by ethanjli