fairscope/PlanktoScope #33: Agitation to keep the sample suspended

fabienlombard, 2020-09-20:
Plankton can easily sink in the sample chamber. A gentle agitation could be provided by bubbles.

gromain, 2020-09-21:
To add more details on this, the problem lies in the sample bottle, not in the imaging chamber. A simple proposition by @fabienlombard is to use a bubbler at the bottom of the bottle to keep the sample homogeneous.
Other systems could be investigated (like a blood pouch agitator), but the simpler, the better.

ethanjli, 2023-09-18:
I am closing this issue because it appears to have been completed: recent versions of the PlanktoScope hardware (since hardware v2.5) now include a bubbler pump to agitate the sample-loading syringe. If that bubbler pump is not sufficient to keep the sample well-mixed before imaging, we can open a new issue for that.

closed 2023-09-18 by ethanjli