fairscope/PlanktoScope #34: Add a way to retrieve the generated pictures other than ssh/ftp/backup drive

gromain, 2020-09-20:
We need to find a way to extract the collected pictures other than having to connect through ssh/ftp to the planktoscope.

I'm thinking about setting up a basic webserver that would list the directory where the zip files generated are located.

This basic webserver could eventually also host the streaming flux from the camera.

gromain, 2020-09-25:
We could also setup a simple http server to download the files when connected to the same network as the planktoscope. One simple file server embedded in python can be started with `python3 -m http.server`.

gromain, 2020-11-29:
Maybe this could be some kind of gallery in Node Red

gromain, 2021-03-18:
Partially solved by 5312f379069749560be00bbd98a855890fafbe32 .

Ideally, there would be a way to navigate to the next picture in the folder, without having to go back to the previous screen.

ethanjli, 2023-08-23:
This issue is completed by #157, which added a deployment of https://github.com/filebrowser/filebrowser to serve the `/home/pi/data` directory with a Google Drive-like browser GUI. As of v2023.9.0-alpha.0, the file browser can be accessed on port 80 at path `/ps/data/browse`. The file browsers functionality includes everything requested in the above comments:
- Any group of files or folders can be downloaded as a .zip or .tar.gz archive (among other supported formats)
- There is a thumbnail view for all images inside a folder
- Double-clicking on an image file brings up an image preview, and it is possible to navigate to the previous and next images in the folder using the keyboard arrow keys.

closed 2023-08-23 by ethanjli