fairscope/PlanktoScope #36: Systematic tests of lenses combination

gromain, 2020-09-21:
We need to improve on two fields:

- depth of focus
- field size

To that end, we need to have a systematic approach to test lens combination and decide on the best compromise.
Fabien reports having tested with a simple macro lens for smartphone as the reverse lens and to have great results, however we currently have no way of measuring the field size.

Ideally, the lens combination is fixed before Tara's test

gromain, 2020-09-28:
@tpollina can you share here the result of your previous testing campaign?

moi90, 2020-09-29:
> we currently have no way of measuring the field size

Don't you have a test pattern?

gromain, 2020-09-29:
I'm not sure what @tpollina used!

closed 2026-11-09 by chevreuill3000