fairscope/PlanktoScope #39: Focus stage is not rigid enough

gromain, 2020-09-21:
The imaging plate is only maintained in position by the two nuts on the stepper focus screws. This is clearly not rigid enough since small movement can happen back and forth. This could lead to a premature failure of the plastic nuts leading to a complete replacement of the stepper assembly.

A proposed low tech alternative is to only have manual control of the focus stage. If we reduce the different flow cells number (see #21) and lens disparity (#36), the interest in having a motorized focus stage is reduced a lot. The stepper assembly could then be replaced by a simple screw/nut assembly with two sliding rods on the side for guiding.

moi90, 2020-09-25:
The addition of two sliding rods would also help to stabilize the stage with the stepper motors in place, isn't it?

gromain, 2020-09-25:
It definitely would do the trick. We just need to put our hand on appropriate rods and sliding bearings!

ethanjli, 2023-10-26:
@tpollina Does hardware v2.5 or v2.6 resolve this problem? If so, I will close this issue as completed.

ethanjli, 2023-11-10:
I haven't heard anything from @tpollina , but I'm pretty sure hardware v2.5 has already resolved this problem (as I believe the plastic lead screw nuts have been replaced by metal lead screw nuts), so I will close it. If it turns out to still be an issue even in the newer hardware versions, we can reopen this issue or open a new one.

closed 2023-11-10 by ethanjli