fairscope/PlanktoScope #40: Design output of PlanktoScope as  "Fit for archive"

rkiko, 2020-09-22:
Raw output format of the PlanktonScope needs to be “fit for archive”, meaning we need a zip file with the images and an according txt file with the metadata (Lat, Lon, Datetime, Depth, Instrument set up, Segmentation algorithm details) that we can submit to a recognised data archive (zenodo, seanoe, pangaea or the like) in an almost automatic fashion. That way, we can submit it, get a DOI and then build the tools to analyse the data on the archived raw material (e.g. import to EcoTaxa and sort it there, but also other data processing will then be enabled). As datasets get a DOI, it will be easy to follow the data.

PlanktonScope should have a 'play' mode and a 'fit for archive' mode. Fit for archive mode will require all metadata set properly (including automatic tests; e.g. lat lon format correct, time correct, e-mail address of data provider, ...)

- need to decide on an archive where to put the data

- need an agreement in the PlanktonScope user group that data is first submitted to archive

- data processing should follow the fair principles: https://www.go-fair.org/fair-principles/

gromain, 2020-09-22:
Thanks @rkiko . To build on your issue, I think the 'play' and 'fit for archive' mode also fits nicely with #32 and the overall settings management.
From what you say, there are at least 3 data archive we should aim to work with:
- Zenodo
- Seanoe
- Pangaea

rkiko, 2020-09-23:
No, we should decide for one archive to go for. Probably, one needs to discuss the needs and data upload strategy with the archive and then decide for the best possible solution/archive. Spreading the data over several archives makes no sense. I think upload should almost be automatic, meaning that if a user sets the "fit for archive" setting, the data is checked and uploaded automatically to the archive (when a proper internet connection exists, otherwise upload in delayed mode). One could also imagine a staging area, where sample acquisitions are loaded, checked by experts for consistency (e.g. random check of images included in the dataset, lat, lon consistent throughout all data submissions from that instrument/person) and then are released to the archive. For CitizenScience, a first check in the staging area might be good. Otherwise, the archive could be flooded with useless data.

moi90, 2020-09-25:
A staging area sounds good but is yet something else that has to be implemented and hosted.

gromain, 2021-07-05:
Since 3f86fb4ddb4ef744f4d1f9d680b2bbc6a2013548 this is done for Ecotaxa.

There is not an automatic upload button (yet) since most machines are used offline. But this could be added quite easily.

closed 2026-11-09 by chevreuill3000

chevreuill3000, 2026-11-09:
Closing as done: as noted above, the "fit for archive" output is implemented for EcoTaxa exports.