fairscope/PlanktoScope #42: Use pi-gen to generate the embedded images in a reproducible way

gromain, 2020-09-28:
For now, master images are created using the golden record method (we create and setup a new raspberry, set everything up by hand and once it works, we pull the image from the sd card and share it).

While this method works, it's not very efficient, quite time-consuming, error prone and leads to big images to be shared with no opportunity for first time setup and such. Setting up [pi-gen](https://github.com/RPi-Distro/pi-gen) could be a good way to streamline the image creation process, to allow for reproducible builds and make released easier and faster.

ethanjli, 2024-04-13:
Update: I've been adding follow-up comments for this requested improvement under #136. Once I try out pimod and/or pi-gen-action, I will report my findings on this issue.

ethanjli, 2024-05-26:
At https://github.com/ethanjli/rpi-forklift-demo I've successfully used Pimod to build a working SD card image with a subset of the setup modules used by the PlanktoScope OS setup scripts (specifically: installation of Docker and a few other CLI tools, and setup+integration of Forklift including OS filesystem overlays). Just like [pi-gen](https://github.com/RPi-Distro/pi-gen?tab=readme-ov-file#how-the-build-process-works), Pimod is just running things in a chroot rather than a booted system, so currently:
- Docker doesn't/can't run in the setup scripts (so `/var/run/docker.sock` doesn't work during setup)
- [The systemd bus is not available](https://0pointer.de/blog/projects/changing-roots) during setup, so we can't start system services during setup

Because of these limitations, Forklift is unable to pre-cache Docker container images, and it's also unable to apply the Forklift pallet specified in the scripts. The next step for Pimod will be to see if we can resolve these limitations. For example, maybe we can use [systemd-nspawn](https://0pointer.de/blog/projects/changing-roots) inside the Pifile to run the setup scripts inside a booted environment (e.g. as discussed at https://forums.raspberrypi.com/viewtopic.php?p=2032673)? If we can't get Docker to run, we may be able to add functionality to Forklift (e.g. using https://github.com/containers/skopeo?tab=readme-ov-file#copying-images) to pre-download all container images to the filesystem during the setup process and then use Forklift to transfer all those container images into the Docker daemon's internal storage during the first boot. Of course, this will make the first boot noticeably slower because it'll have to create a bunch of new containers.

I also found that Blue Robotics's BlueOS (a Raspberry Pi OS-based OS) also uses Pimod to generate their SD card images: https://github.com/bluerobotics/BlueOS/tree/master/deploy/pimod . [They use Docker-in-Docker](https://github.com/bluerobotics/BlueOS/blob/master/install/install.sh#L134) to run Docker-related setup script commands when they run their setup scripts in CI.

ethanjli, 2024-05-27:
Update: in https://github.com/ethanjli/rpi-forklift-demo/pull/1 I've successfully automated the build process of https://github.com/ethanjli/rpi-forklift-demo (which uses various setup scripts copied from PlanktoScope OS's setup scripts) without using Pimod/pi-gen. Specifically, I run almost of the setup scripts in an unbooted `systemd-nspawn` container as a cleaner/simpler alternative to Pimod's chroot-based system, and then I run one more setup script (which interacts with the Docker daemon, which must run in a booted OS) in a QEMU virtual machine. The resulting SD card image is processed in GitHub Actions with [pishrink](https://github.com/Drewsif/PiShrink). This demonstration solves the problems described in my above comment, and it provides a clear path to fully automating PlanktoScope OS's SD card image build process inside GitHub Actions.

ethanjli, 2024-06-06:
In PR #416 I've set up automated OS builds using the same setup scripts as what we use for user-initiated setup (using https://github.com/PlanktoScope/install.planktoscope.community/blob/edge/distro.sh), by running those setup scripts in a combination of systemd-nspawn containers (for almost everything) and a QEMU virtual machine (for talking to Docker, though I'd like to remove the need for the QEMU VM because it's really slow - that setup step takes ~30 min to run and is a source of flakiness).

closed 2024-06-07 by ethanjli

ethanjli, 2024-06-15:
PR #428 significantly speeds up the automated OS builds (by ~20 min), so now each OS image takes ~40 min to build!

PR #416 (2024-06-07): Start building 64-bit OS images in GitHub Actions
This PR closes #42 by setting up a GitHub Actions workflow to build OS images in GitHub Actions, using a combination of https://github.com/marketplace/actions/run-pinspawn and https://github.com/marketplace/actions/run-piqemu to run the PlanktoScope OS setup scripts on Raspberry Pi OS images.

- Currently each image takes 1 hour to build and then an additional ~20 minutes for me to download, which is comparable to (or slightly longer than) the amount of time I had needed to spend on manually building each image; but because Github Actions allow [... cut]