fairscope/PlanktoScope #44: Precise the necessary and nice-to-have metadata list

gromain, 2020-09-28:
We should list the metadata we want to save during a sampling and during an image analysis.

Top of my head:
- gps position
- date and time of sampling
- sampling volume
- sampling duration
- sampling speed
- sampling device

I'm probably missing a lot, please add to the list!

moi90, 2020-09-29:
Apart from datetime and GPS, all information has to be provided by hand, right?

Probably much more important than the volume of the whole sample is the *imaged* volume that is visible in one full frame. Unfortunately, this depends on a lot of things and people won't be able to give that information accurately.

gromain, 2020-11-29:
Here is the exported metadata, for now:

```
sample_project
sample_id
sample_ship
sample_operator
sample_sampling_gear

acq_id
acq_instrument
acq_instrument_id
acq_celltype
acq_minimum_mesh
acq_maximum_mesh
acq_min_esd
acq_max_esd
acq_volume
acq_magnification
acq_fnumber_objective
acq_camera_name
acq_nb_frame

object_date
object_time
object_lat
object_lon
object_date_end
object_time_end
object_lat_end
object_lon_end
object_depth_min
object_depth_max

process_pixel
process_id
```

`object_date_end`, `object_time_end`, `object_lat_end`, `object_lon_end` are only exported when the selected sampling device is a net.

gromain, 2020-11-29:
I'm not sure when the `process_id` should change, I think I'll set it to 1 for now.
If I'm not mistaken this should change when the raw images are processed by the segmenter.

gromain, 2020-11-29:
Also, missing today is the imaged volume.

gromain, 2020-11-29:
So I've added `acq_imaged_volume` metadata. Calculated from the camera type and manual field measurement.

For reference, with the HQ camera, the field is 4,15mm x 3,14mm and the resolution 1µm/px.
With the camera v2.1, the field size 2,31mm x 1,74mm, resolution 0.7µm/px.

Obviously, this depends on the lenses used, and for now this only works for the 25mm and 16mm couple.

tpollina, 2020-11-29:
The `process_id` is defined by MorphoCut when it's doing a segmentation.

gromain, 2020-11-29:
Ok, I'll removed the field from the segmenter tab then!

gromain, 2020-11-29:
I'll also add a
> `sample_volume_sampled` (ie the volume filtered by the net) that could be alternatively calculated from lat/lon start end and net opening and a `sample_concentrated_sample_volume` (ie got from the collector). Without those quantitative samples will get impossible
> And therefore `sample_gear_mesh_size`
> And `sample_gear_net opening`
> Are needed too
from @fabienlombard

ethanjli, 2023-11-22:
@LaurentPV I just read this issue for the first time, and it sounds directly relevant to what you're working on now with https://docs.google.com/spreadsheets/d/1TSIaOFEIMvvYyqAFrsiZxVtGXZvWVdZbWO_LU-2A_TE/edit . So I've assigned this issue to you.

LaurentPV, 2023-11-22:
A table with all the informations regarding currently used and to be created metadata variables was created [here].(https://docs.google.com/spreadsheets/d/1TSIaOFEIMvvYyqAFrsiZxVtGXZvWVdZbWO_LU-2A_TE/edit?usp=drive_link)

Please feel free to explore this document if you have access and keep the comments in this issue.

ethanjli, 2023-11-23:
I'd like to close this issue earlier rather than later - so @LaurentPV after you look over all the fields proposed by this issue and, for each field, either confirm that it's added to https://docs.google.com/spreadsheets/d/1TSIaOFEIMvvYyqAFrsiZxVtGXZvWVdZbWO_LU-2A_TE/edit?usp=drive_link or that you you don't intend to add it, let's close this issue.

LaurentPV, 2023-11-23:
You can close it, everything is on the metadata compilation file.

closed 2023-11-23 by ethanjli