fairscope/PlanktoScope #51: Stabilize the imager script

gromain, 2020-10-03:
If we have a dirty shutdown of the imager script (via a `sudo killall -9 python3` for example), the script sometimes doesn't start and fails silently on the camera instantiation. The only way to salvage the situation is to do a full restart of the machine.
So yeah, not good.

One way to manage this could be to capture signals in all the sub-scripts. Another way could be to wrap the camera api calls in `try ... except` wrappers.

moi90, 2020-10-03:
SIGKILL (9) stops the process immediately without allowing it to clean up. SIGINT, however, is converted to KeyboardInterrupt. I suspect that when a process is killed with SIGINT, the camera properly de-initializes, is that right?

gromain, 2020-10-03:
That is what should happen. Except it's not happening everytime.
Sometimes, if the sub-process is stucked somewhere, it won't catch the event and stay in its loop.
I think I'll add a way for the sub-processes to catch a signal directed to them.

moi90, 2020-10-05:
Are we using python multiprocessing? Then, it gets messy... If I recall correctly, it is not guaranteed that all processes receive a signal. For example, if the parent receives SIGKILL and the child does not, the child will keep running. So the child needs to check periodically if its parent is still there.

gromain, 2020-10-05:
In the new version of the python code, it helps to decouple the logic between the different elements.
Right now, I pass an event to every child that gets asserted when the main process receives a SIGTERM.

This works really well, except when the imager thread locks up. I've had a look at the picamera code and it may be coming from a wait for a lock. I have to look more closely though. I think the problem may be coming from the fact we are using a subprocess to run the camera. This should not be a problem since the camera is created inside the subprocess. But I may have a small tiny thing somewhere that locks the camera up.

closed 2020-11-16 by gromain