fairscope/PlanktoScope #54: Camera failure on capture

gromain, 2020-10-03:
Sometimes, during an imaging session, the camera will simply not capture and will wait forever the capture to happen.

This happened two times, once on the third picture taken, the second after ~800/1000 pictures.

Maybe we should stop the streaming server before starting the capture?

gromain, 2020-10-03:
This is possibly linked to #51 .

gromain, 2020-10-03:
Happened again on pic 136/175... Getting more elusive by the second!

moi90, 2020-10-03:
Yes, I think the streaming should be stopped during capture, because I guess it would also interfere with the flash (if that turns out to be useful). Instead, the pipeline could send a downsampled version of the captured frame to the frontend. (And even enhanced versions with highlighted objects.)

gromain, 2020-10-03:
My guess too. I'll test it and see how it goes!
The enhanced version would definitely be nice to do!

gromain, 2020-10-19:
Right now the work on this goes is centered around replacing the [condition used for synchronisation](https://github.com/PlanktonPlanet/PlanktonScope/blob/master/scripts/planktoscope/imager.py#L46).

This primitive is used to synchronise the availability of a new frame and its delivery to the streaming server (and subsequently to the user's browser). This primitive causes problem if we try to shut down the camera right at the moment the frame is sent to the server.

A way to replace this synchronisation is to use a queue or a pipe to send the frame to the streaming server. The preliminary tests works well, I just need to solve limit issues around proper close of the pipe. I haven't been able yet to reproduce the error when using this setup, but I'm not sold yet on the fact that this is solved definitely.

moi90, 2020-10-19:
I wouldn't recommend a Pipe as it is more low-level (and for example can block mid-message if the OS buffer is full. a Queue doesn't do that). Although a Queue can be infinitely large (maxsize=0), it then stores all items indefinitely, which is certainly not what we want. If maxsize>0, the queue can block the caller, which is also not what we want.

Can you further explain the problem with the Condition? In my understanding, a Condition should work best.

closed 2020-11-16 by gromain