fairscope/PlanktoScope #60: Add a way to customize the segmentation parameters

gromain, 2020-10-21:
Parameters for the segmentation are hardcoded. However, some objects require different settings. There should be a way to change the parameters of a standard pipeline run.

gromain, 2020-11-29:
The segmentation will not be done on the machine on Tara, so I'm getting this issue out of the Tara milestone now.

gromain, 2021-12-01:
Regarding the UI, in the segmentation tab:
- [ ] Add a slider to specify the minimum object to segment (default set to minimum fraction size sampled)
- [ ] Add a slider to specify the maximum object to segment (default set to the maximum fraction size sampled)

ethanjli, 2024-04-04:
Update: We discussed this improvement (which was also requested by @fabienlombard ) in the [2024-04-04 software meeting](https://docs.google.com/document/d/1Iba0_9yNFEDdp1Qy8IYlLP43Ga0DnDr3CdavWV0QGZY/edit#heading=h.ydmdu4ns30fq), in the context of @BiyouS 's project to rewrite the Node-RED dashboard. Currently it seems that there are too many unknowns (about the segmenter settings we'd like to expose) and the potential complexities involved are too high, so it will be out of the scope of @BiyouS 's initial rewrite. The plan is to leave design of a proper GUI for segmenter setting as work to be done in a future iteration of the GUI.

closed 2026-11-10 by chevreuill3000