fairscope/PlanktoScope #61: Better management of sample_id, acq_id and process_id

gromain, 2020-10-21:
This issue is closely linked to #44 .

Right now, we have a semi-automatic system in place of management of ids.
`sample_id` is a number incremented by one on startup, the rest is static and user configured.
However, users commented that sample_id could also be a name, date, etc...
So this field should not be a number only that is incremented automatically.

`acq_id` and `process_id` on the other hand could be incremented automatically, that is if a capture of a particular sample already exists, the `acq_id` could change. Or better yet, we should ask the user what to do.

Same goes for `process_id` even if for now we don't have a way to change the segmentation settings in the UI (this is tracked by #60).

moi90, 2020-10-25:
I would additionally include a hardcoded, not resettable ID that is incremented / uniquely generated for each run and therefore without doubt identifies each run (even if the other IDs are sometimes messed up). Errors happen and this allows to track runs even if the other IDs are changed (to fix errors etc).

gromain, 2020-11-24:
This was improved in part by #67 .

gromain, 2020-11-29:
So now these id are not automatic anymore and are managed solely by hand.

Eventually, I would like to add a system where the last `acq_id` used is shown for the chosen specific `sample_id` if this one exists. Same thing for the `process_id`.

moi90, 2020-11-30:
Still, I find it very important to have an additional ID that users can not mess with. Errors do happen.

gromain, 2020-12-02:
I've added what you suggested @moi90 . There is now a `sample_uuid`, `acq_uuid`, `process_uuid` that are added everytime a new process is run

moi90, 2020-12-02:
Ah, great!