fairscope/PlanktoScope #69: Can't access `planktoscope.local` from an Android phone

gromain, 2020-11-25:
And more generally, mDNS is not very well supported by Windows either.

This is a problem in Android itself, so definitely a wontfix here, but we need to find a way for this to work for our users anyway.
One way to do so could be a dedicated app. This has a lot of advantages, but is a major PITA.

Upstream bug is there, but it's not likely something will happen anytime soon: https://issuetracker.google.com/issues/140786115

ethanjli, 2023-05-25:
@gromain #137 configures the Raspberry Pi to act as a DNS server serving a DNS record associating "home.planktoscope" with `192.168.4.1`. I confirmed with testing from my Android phone that the DHCP process (when the phone connects to the Raspberry Pi as a wireless AP) configures the Android phone to use the Raspberry Pi as a DNS server. However, Google Chrome on Android only uses the Raspberry Pi as the DHCP-configured DNS server when the "Privacy and security -> Use secure DNS" setting is set to "Use your current service provider (Secure DNS may not be available all the time)" instead of "Choose another provider". Would you consider this an adequate resolution for this issue? Otherwise, should we close this issue with a "wontfix" label?

ethanjli, 2023-09-18:
It looks like Android has [added](https://www.esper.io/blog/android-dessert-bites-26-mdns-local-47912385) mDNS support. I have not tested this yet.

ethanjli, 2023-09-18:
I just tried opening `pkscope.local` on my Android phone (Google Pixel 5A) with the v2023.9.0-beta.1 prerelease, and the landing page successfully loaded! However, it's likely that there are still plenty of Android phones out there which have not yet received the update adding mDNS support. Because mDNS support on Android is now in the pipeline without any additional effort required by us, and because #137 adds a viable alternative (which, as of #214, has been shortened to `home.pkscope`) for many devices - as long as they aren't using Private DNS - I think we can consider this issue resolved. However, I will leave the docs in their current form for now, since mDNS support on Android is probably still not yet widespread beyond Pixel phones:

> If you connect another device (e.g. a phone or computer) to the PlanktoScope's Raspberry Pi over its local isolated wifi network or over an Ethernet cable, then you can open a web browser on the device to access the PlanktoScope's dashboard at one of the following URLs (try them in the following order, and just use the first one which works):
>
> - http://home.pkscope/ (this should work unless your web browser is configured to use a Private DNS provider)
> - http://pkscope.local/ (this should work unless you're on a device and web browser without mDNS support; notably, Google Chrome on Android does not have mDNS support)
> - http://192.168.4.1/ (this should always work)
> - http://192.168.5.1/ (this should always work)

closed 2023-09-18 by ethanjli