fairscope/PlanktoScope #74: Sampling Gear - Add culture

tpollina, 2021-03-18:
Having the capacity to select the **culture** as a proper **samping gear** is needed

In this case, sample location could be filled either automatically by the GPS if it's active or stay blank (it's painful to fill them when you don't need them for the study)

gromain, 2021-03-18:
I beliebe the location metadata is necessary for ecotaxa though. So we may deal with this requirement either with:
- not creating an ecotaxa archive on segmentation if there is no GPS data (and let the user know why the archive is not created), but still do a segmentation
- automatically filling a GPS location (if there is a GPS hat connected, with proper satellite reception)
-

gromain, 2021-12-01:
Choice has been made to force entering dummy GPS coordinates. I suggest using the south Pole, since there is no ocean there.

Regarding the UI, changes need to be made in Sample tab:
- [ ] Add a culture in sampling gear. No need to ask for the GPS location for it (force insert dummy GPS coordinates)
- [ ] Add a test in sampling gear. No need to ask for the GPS location for it (force insert dummy GPS coordinates)

gromain, 2021-12-03:
closed by 8279ab0

commit 8279ab02 (2021-12-03): flow: add culture and test
Closes #74
commit 0e3a71b3 (2021-12-03): flow: add culture and test

Closes #74
closed 2021-12-03 by gromain