fairscope/PlanktoScope #75: Concentrated sample volume

tpollina, 2021-03-18:
The input **Concentrated sample volume** stays active when the sampling gear **Niskin bottle** or **Pass Hull** are selected.

It could make more sense to remove this input in this case

gromain, 2021-03-18:
Can a Niskin bottle sample be concentrated before imaging though?

tpollina, 2021-03-18:
Effectively, the Niskin bottle sample can be concentrated before imaging.

ethanjli, 2023-11-22:
A Niskin bottle sample can be concentrated before imaging (which I did a lot this past summer), and in general it does need to be concentrated before imaging unless the sample is collected from an extremely heavy bloom. So I think we should keep the "Concentrated sample volume" field around as an optional field in the Niskin bottle sample type (unless I am misunderstanding the meaning/purpose of that field).

I don't know what a "pass hull" sample means - is that underway water, as in water which is continuously pumped from below the ship's hull while the ship is moving? If so, such samples also generally need to be concentrated before imaging in the PlanktoScope, so I think it would make sense to keep a field for recording the concentration factor.

If my understanding is correct, we should not implement the change proposed by this issue. Unless someone provides additional information in the next month supporting the removal of the "Concentrated sample volume" field for the Niskin bottle and "pass hull" sample types, I will close this issue.

chevreuill3000, 2026-11-11:
Closing as not planned: the discussion above concluded that a Niskin bottle or underway sample is usually concentrated before imaging, so the volume fields have to stay available for these sampling gears. The pending interface rework keeps them for the Niskin bottle and hides them only for lab cultures and demo samples. Please reopen if there is a case where the field is misleading.

closed 2026-11-11 by chevreuill3000