fairscope/PlanktoScope #76: Automatize the white balance calibration

tpollina, 2021-03-18:
The White Balance needs to be manually calibrated via the channels Red and Blue using the color of the streaming under the Auto White Balance.

It could be nice to have this step automatized in order to standardize the color calibration over different samples and different machines.

gromain, 2021-06-18:
This needs to be modified in raspimjpeg so that we can retrieve the values set by the awb algorithm of the camera.

See https://www.raspberrypi.org/forums/viewtopic.php?p=569756#p569756 and https://github.com/waveform80/picamera/commit/1fbe0032ac9f43bd9970303dc01e26d578f6afb4 .

ethanjli, 2023-09-06:
Closing this in favor of #204, which is a duplicate feature request but proposes a way to automatically calibrate the manual white balance gains without having to rely on raspimjpeg (which we anyways want to throw away in favor of libcamera+picamera2). That method will work independent of the specific camera library, which I think is a good thing to make it easier for us to eventually [delete](https://programmingisterrible.com/post/139222674273/write-code-that-is-easy-to-delete-not-easy-to) code related the underlying camera driver, whether that's raspimjpeg or libcamera or whatever NVIDIA Jetson camera drivers.

closed 2023-09-06 by ethanjli