fairscope/PlanktoScope #77: Brightness of the pictures taken seems different between v2.2 and v2.1

gromain, 2021-03-23:
In v2.1, users reported that with the camera v2.1 and the following settings the brightness of the picture was enough:

- ISO: 60
- Shutter speed: 100µs
- Exposure mode: off
- awb mode = off

Under v2.2, with the same settings, the brightness is much lower and pictures are under exposed.

gromain, 2021-03-23:
Between the two versions, we switched from using `picamera` to `raspimjpeg`. The problem can lie either in the way `raspimjpeg` is setting the camera up, or in something `picamera` was doing that we are not doing now.

ethanjli, 2023-10-26:
I'm closing this issue as "wontfix" because I'm not sure anyone is using v2.2 anymore, and I think we're currently just living with whatever raspimjpeg is doing now, if this issue hasn't already been fixed. We are anyways planning to migrate from raspimjpeg to the picamera2 library.

closed 2023-10-26 by ethanjli