fairscope/PlanktoScope #80: Motor Hat Import Board permission issue

arhyneRWU, 2021-05-04:
**Describe the bug**

The Stepper motor fails to work because of a permission issue, it can not import the module 'board'
```
Log cleared!
2021-05-04 16:20:04.367 | INFO | __main__:<module>:39 - Starting the PlanktoScope python script!
2021-05-04 16:20:04.451 | INFO | planktoscope.mqtt:<module>:60 - planktoscope.mqtt is loaded
Traceback (most recent call last):
File "/home/pi/PlanktonScope/scripts/main.py", line 45, in <module>
import planktoscope.stepper
File "/home/pi/.node-red/projects/PlanktonScope/scripts/planktoscope/stepper.py", line 3, in <module>
import adafruit_motorkit
File "/usr/local/lib/python3.7/dist-packages/adafruit_motorkit.py", line 55, in <module>
import board
**File "/home/pi/.local/lib/python3.7/site-packages/board.py", line 166, in <module>
raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None**
<13>May 4 16:28:00 update.sh: Updating the main repository
<13>May 4 16:28:00 update.sh: nothing to do!
```

**To Reproduce**
Steps to reproduce the behavior:
1. ssh into the pi
2. cd test
3. python3 stepper.py
`File "/home/pi/.local/lib/python3.7/site-packages/board.py", line 166, in <module>
raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None `

4. sudo python3 stepper.py  <---- this works without the error

**Planktoscope version (please complete the following information):**
Pi4, Verison 2.2,  Adafruit Hat

gromain, 2021-05-04:
Hey, thanks for reporting the issue!
You are using the provided SD card image right? Is it not working only on the test script or also in the main application (after a fresh reboot for example)?

That's super weird anyway.

@tpollina can you try to reproduce, I think you have machines assembled with the Adafruit hat?

gromain, 2021-05-04:
Also, it's the official Adafruit stepper hat right? (just making sure)

arhyneRWU, 2021-05-04:
@gromain thanks for the quick reply.  
This is the log from the main application, via the node-red HTTP interface. 
`Log cleared!
2021-05-04 16:20:04.367 | INFO | __main__:<module>:39 - Starting the PlanktoScope python script!
2021-05-04 16:20:04.451 | INFO | planktoscope.mqtt:<module>:60 - planktoscope.mqtt is loaded
Traceback (most recent call last):
File "/home/pi/PlanktonScope/scripts/main.py", line 45, in <module>
import planktoscope.stepper
File "/home/pi/.node-red/projects/PlanktonScope/scripts/planktoscope/stepper.py", line 3, in <module>
import adafruit_motorkit
File "/usr/local/lib/python3.7/dist-packages/adafruit_motorkit.py", line 55, in <module>
import board
**File "/home/pi/.local/lib/python3.7/site-packages/board.py", line 166, in <module>
raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None**
<13>May 4 16:28:00 update.sh: Updating the main repository
<13>May 4 16:28:00 update.sh: nothing to do!`

I ssh-ed to confirm the bug and see if it was a permission issue.  Both the issue at the test/stepper.py and the node-red application seem to be identical.

arhyneRWU, 2021-05-04:
> Also, it's the official Adafruit stepper hat right? (just making sure)

We ordered directly from the excel sheet posted on the website, but I can confirm this with my student when I see him next. I'm tempted to just change the permission on the modules to allow any user access to it and see if that solves the issues but wanted to report it out first.

gromain, 2021-05-04:
And you flashed an official image, you didn't follow the expert route right?
I think it's the first time we have this issue, so very strange it appears!
We'll get to the bottom of it, but yeah, in the meantime, just changing the permissions on the modules could (should!) work.

arhyneRWU, 2021-05-05:
I'm confirming that Kyle (undergrad I have working on this build) installed the image V2.2: released 2021-01-15.  I get called in when there is a need for help, a question, or an issue.    I'm certain he installed the official image and I don't think he would have installed an older version as he is following the install manual exactly. 

I'm going to reimage it just to confirm the permission issues.  I wrote some tests trying to find the errors, so a fresh image will clean out my scratches. 

One other note, the camera is not streaming on the UI Optic Config.  It has the broken image link icon.  The camera works as I tested it directly on the pi.  My assumption is that this issue is tied to the permission issues we have with the motor hat.  The GPS and Fan are working properly.

gromain, 2021-05-06:
I was just checking all the bases. This problem is definitely not common and really unexpected! We have other machines working flawlessly with the same image and the same hardware. I'll see with Thibault if he had time to try to reproduce the error

It's normal that the camera is not working, it's managed by the python script, and the crash on startup make this fail.
This is something I'm going to address in the next release, making sure that if a module fails, other still works.

arhyneRWU, 2021-05-07:
@gromain be good to know exactly what RasPi you are all using. I dont' know if it is hardware or not.   I can get the stepper hat to communicate with me using its test. I can get main.py to work to the point of failing at 'import morphocut' by, 'sudo python3 main.py '

I was able to test the camera out with a standalone python script to confirm it works.

arhyneRWU, 2021-05-07:
I'm going to take the system home this weekend and tear it apart and test the motor driver separately.  I might order another one just in case.

arhyneRWU, 2021-05-10:
I installed a new stepper hat, however, the error remains.  It is a permission issue.  

I'm not sure why this has popped up, hardware? operating system?  Node-RED version? 

I can provide complete log **journalctl -b**  if that is helpful

I have updated the Node-RED flow to run the python under sudo which allows the _main.py_ script to successful run, but other process are still broken.   Camera stream URL doesn't not work but the camera was started and works from looking at the logs.  If I kill the camera and run the camera in test script it run and is accessible.  

I'll move the discussion to SLACK.

arhyneRWU, 2021-05-10:
@gromain I have the board.py issue fixed

my PI is a bit different than some others I guess. 

root was a few version ahead of pi with respect to adafruit-platformdetect

pi was v2.14  root was v2.17  

I'm not sure if the most recent version is compatible with the system, but this is what I ran. 
`pip install --upgrade adafruit-blinka adafruit-platformdetect`

babo989, 2021-05-10:
Seeing the same error with a newer Pi 4b
"Log cleared!
2021-01-15 13:52:17.432 | INFO | __main__:<module>:39 - Starting the PlanktoScope python script!
2021-01-15 13:52:17.486 | INFO | planktoscope.mqtt:<module>:60 - planktoscope.mqtt is loaded
Traceback (most recent call last):
File "/home/pi/PlanktonScope/scripts/main.py", line 45, in <module>
import planktoscope.stepper
File "/home/pi/.node-red/projects/PlanktonScope/scripts/planktoscope/stepper.py", line 3, in <module>
import adafruit_motorkit
File "/usr/local/lib/python3.7/dist-packages/adafruit_motorkit.py", line 55, in <module>
import board
File "/home/pi/.local/lib/python3.7/site-packages/board.py", line 166, in <module>
raise NotImplementedError("Board not supported {}".format(board_id))
NotImplementedError: Board not supported None"

gromain, 2021-05-10:
I'm going to try tonight on my machine. 

But what's weird is that I'm running with the latest Raspberry and I had no problem with the image 2.2 (since I created it on my machine).

@babo989 I think you have another problem. Is the Adafruit motor board connected when you try to start the machine?

arhyneRWU, 2021-05-10:
It's the Pi itself, it's a hardware conflict (newer Pis) with the software board.py, _adafruit-platformdetect_ is looking at the type of chip on the Pi, newer Pi has a different chip which is not listed in the board.py library.   It is basically telling the motor board which pins it can use.  So it doesn't find info and errors. 

I can make a map of the process if you wan to see it.

arhyneRWU, 2021-05-10:
> It's the Pi itself, it's a hardware conflict (newer Pis) with the software board.py, _adafruit-platformdetect_ is looking at the type of chip on the Pi, newer Pi has a different chip which is not [...]

I agree that sudo is not a good option. I was just debugging and notice the library difference between sudo and pi, which pointed me to the correct problem.  

pip install --upgrade adafruit-blinka adafruit-platformdetect

Fixes the issue.  It updates the board.py for newer hardware.

gromain, 2021-05-10:
I'm trying now, I'll release a bugfix image tonight if I can reproduce.