fairscope/PlanktoScope #81: Make modules independent from one another

gromain, 2021-05-06:
Right now, a failure of any of the python process leads to a complete stop of everything. While this was a feature, it's a considered a bug by many users, and it's actually a desirable behaviour for tests to be able to run the machine with partially setup equipment.

Thus, we need to decouple the python module from one another and make sure that the crash (or non startup) of one doesn't affect the others and don't lead to a shutdown of `main.py`.

Eventually, we should find a way to deactivate the controls of a specific module in Node-Red (for example if we receive a "Dead" message from one module).

ethanjli, 2024-03-26:
Update: #187 (included in the v2023.9.0 software release) made the segmenter fully isolated from the hardware controller processes; now the segmenter is launched as a separate system service (launched/supervised by systemd in v2023.9.0, launched/supervised by Docker starting in v2024.0.0). #380 changes the default imager process to a new (picamera2-based) version which does not cause the backend to quit even if an error (e.g. missing camera) occurs; instead, the imager process proactively cleans up its internal state and then idles until it receives a shutdown signal, at which point it finally quits. That could be simplified by changing the `main.py` script of the Python backend to not shut down other processes when one process is detected to have shut down - which is the improvement proposed by this PR. I will work on implementing this change as part of the incremental rewrite of the Python hardware controller.

ethanjli, 2024-04-04:
@Oumayma-hy This change will be part of our incremental rewrite of the hardware controller, and it should probably be relatively straightforward to implement this change.

chevreuill3000, 2026-11-09:
Closing as done: the hardware controller is now split into one systemd service per component (pump, focus, light, imager, bubbler, display), so a failure in one process no longer affects the others.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-11-09 by chevreuill3000

chevreuill3000, 2026-11-09:
Closing as done: the hardware controller is now split into one systemd service per component (pump, focus, light, imager, bubbler, display), so a failure in one process no longer affects the others.