fairscope/PlanktoScope #88: Rewrite the whole updater

gromain, 2021-07-05:
Right now, the updater doesn't work properly. Using git for this is a bad idea(tm).

We should rewrite the whole bunch, either starting with readonly fs and binary delta distribution, or some other form of version management.

ethanjli, 2021-12-10:
Do you have a summary of use-cases and requirements for the updater, and/or how the current updater script is causing problems?

If you want to make the SD card read-only in order to avoid write degradation of the SD card, e.g. by enabling overlayfs configuration (which I've found is a rather involved process to configure, but can be [fully scripted](https://github.com/pez-globo/pufferfish-software/blob/develop/deploy/scripts/overlayfs.sh)), but you also need to update planktoscope software without flashing a new image onto the SD card, then you'll need an external writable media anyways (e.g. a USB drive). Then it'll probably be simplest to just store all the software on the writable media, and then delete+replace the directory containing that software as part of the update process. Depending on requirements and priorities, this could be done with something like cURL + unzip/tar/etc. You could have a "bootstrap" installer script on the read-only FS which downloads the actual installer script anywhere and runs it, and then the actual installer script downloads the rest of the software files.

Updating any system configuration files as part of a software update on a system with overlayfs is complicated, but I believe it's possible: you'd have to disable overlayfs (I know this can be done from the RPi settings interface, so it should probably be scriptable; but I've never tried to script it), then you might need to restart the system and make any necessary changes, and then you'd need to enable overlayfs and restart the system again. It'd probably be simpler to just re-flash the SD card with a new image with the updated system configuration, if/when that's feasible.

gromain, 2021-12-12:
This is something we need to define precisely.

I think the two most common use cases are:
- machine connected to the internet
- machine isolated (need to use a USB drive to bring the update package)

In terms of what we want to update, there are two cases:
- Whole OS, libraries and associated configuration
- PlanktoScope package only

What I had in mind was setup a read only filesystem with 3 partitions, one read only for the base os and libraries and two read-write partitions, one for data and machine logs, one for planktoscope software package.
I was thinking about using something like [swupdate](https://sbabic.github.io/swupdate/index.html), which along with an [hawkbit](https://www.eclipse.org/hawkbit/) server to manage machine versions. What's nice with swupdate is that it can detect if a usb drive brings a software update package which can be quite helpful for machine with difficult network access.

The current updater script is [there](https://github.com/PlanktonPlanet/PlanktoScope/blob/master/scripts/bash/update.sh). It was using git as base for the PlanktoScope update, but was a pain to manage when files where changed inside the folder (for config files for example). The script also had pre-install and post-install section, but that too was a pain to manage in case the machine config had a tiny slight difference from the one the script was created on. In the end, I deactivated completely this function.

ethanjli, 2024-04-13:
I'm going to close this issue, as I consider it superceded by #133. I'm closing it because I've made enough progress on #349 that I am now very confident in the feasibility of the approach described in #133. Future updates will be tracked in #133 and its sub-issues.

closed 2024-04-13 by ethanjli