fairscope/PlanktoScope #89: Dynamic segmenter minimal size selector

gromain, 2021-07-05:
Right now, there is no way to change the minimal size of the objects found by the segmenter.

There should be a slider somewhere to set this.

gromain, 2021-08-11:
Partially solved for now, uses min mesh size * 2.
No dynamic selection though.

ethanjli, 2024-01-26:
I'm adding this feature request to the backlog because @fabienlombard [requested](https://planktoscope.slack.com/archives/C01V5ENKG0M/p1706296112580049) it today in a discussion about the segmenter. This feature request will need some thoughtful design about how to represent any default or non-default size threshold in the metadata, because this is an important experimental parameter which determines the composition of the data produced by the segmenter (just like mesh filter size determines the composition of objects imaged by the PlanktoScope).

ethanjli, 2024-10-10:
This feature request was discussed again in detail in the [2024-10-10 software meeting](https://docs.google.com/document/d/1Sq0rqxOLi0h2hlvAo50e_UXm2fOa1FaKFnn4CNLB8TI/edit?tab=t.0#heading=h.7amegkl1va0x). @tpollina has made a tentative name for the metadata field to represent the size threshold (though this might be changed): `process_min_ESD`. @fabienlombard might be the person to make sure we have discussions in future meetings to make design decisions for this feature's UI/UX, e.g. whether users should set the value in units of pixels or um; and other design decisions which we haven't thought about yet.

ethanjli, 2024-10-10:
As part of a DM conversation with me on Slack, Fabien says the minimum size threshold should instead be calculated as follows:
```
minESD = minMesh    # or process_min_ESD in future  (in micron here)
        pixel_size = self.__global_metadata.get(“process_pixel”, 1.0)
       minradius_pix = minESD /2 /pixel_size  # now we work in pixels
 minArea = math.pi * (minradius_pix ) * (minradius_pix )
```

ethanjli, 2024-10-10:
Here is some discussion on design decisions I had with Fabien:

me:
> Here are a few design decisions we will need to make for the UI:
> - The name of the "process_min_ESD" setting which we will show to users on the segmentation page
> - Should the setting be adjusted as a slider (like ISO), or a text input box (like the "Flowrate" setting), or a number incrementing box (like "Delay to stabilize image" setting)?
> - What should happen if the user does not know what value to use for the setting? Should we provide a fixed default value which the user can just accept, or should we leave it blank by default but make it a required setting (so that Node-RED will require the user to enter some number), or should we leave it blank by default as an optional setting and then make the backend automatically decide for the user?

Fabien:

> > The name of the “process_min_ESD” setting which we will show to users on the segmentation page
> 
> = minimal size to segment (ESD in micron)
> > Should the setting be adjusted as a slider (like ISO), or a text input box (like the “Flowrate” setting), or a number incrementing box (like “Delay to stabilize image” setting)?
> 
> text input box (prefilled to 20 could be an idea)
> if we can reload the previous usage filled by the user : even better
> (prefilled by 20 at first usage, if the user use a 63 micron mesh: he do not whant to change this value everytimes)

chevreuill3000, 2026-11-11:
This request is implemented: the Segment dialog asks for the minimal area-equivalent diameter in µm (default 20) and, since #916, the segmenter converts it to pixels with the pixel size of the acquisition. The check of that conversion on an instrument is tracked in #899, and showing the value used for each segmentation in #1080. The pending interface rework (#1088) keeps this field. Please reopen if the threshold can no longer be chosen before a segmentation; a wrong size in an export belongs in #899.

closed 2026-11-11 by chevreuill3000