fairscope/PlanktoScope #90: Various UI improvement tips

gromain, 2021-07-05:
On the sample page
- add an (optional) field for “concentration factor” ( “datatip” fill 0.5 if dilluted by two; fill 2 if concentrated by a factor 2)
- modify as an (optional) field the “concentrated volume” ( add as datatip “i.e. volume obtained by the net cod end”)
- add an (optional) field for “filtered volume” ( datatip ” total volume of water filtered by the net”)
- add an (optional) field for “net opening diameter” ( datatip ” diameter of mouth of the net”)
- add an (optional) field for “net mesh size” ( datatip ” mesh pore size”)
- add 3-4 “free” fields with two part: a free “label” and a free “larger cell” (pre-fill the first one with the label “comment” and a statement “enter comment here”

On top of the page:
- add a button to “recall” previous enters (everything except lat/lon/time/ and all the proposed volumes/concentrations factors and free fields but keep the “net informations)

All those should be transmitted to the metadata.json up through ecotaxa (as sample_XXX, free fields should become sample_“label” with the first one being sample_comment (caution, it should be intelligent to filter out the “enter comment here” placed as an example

On the optic configuration page:
- reorganize the different parts with this order: optic characterization/ camera settings/ preview/ focus adjustments / fluidic manipulations (this should allows better and convenient focus as well as adjusting camera settings)
- Add a button to take “free capture” and “free video”
- Add a button to recall previous camera settings to the “camera setting page”

- automatic white balance should be a “push-but auto-release” button and should lead to an update of the WB:red and WB: Blue sliders (ie. there is currently no possibility to ensure correct White balance and the “auto”white balance ” should not have the possibility to change during acquisition). Linked to #76 , needs a feature addition to raspimjpeg to add the possibility to retrieve the auto set value.

On the fluidic acquisition page:
- add a button to recall previous options (number of images/pumped volume/delay between images)

On the segmentation page
- add an acces to some options of the segmenter (parameter of the adaptative thresholding, radius of the kernel for erosion/dilatation)

you can eventually hide this in a “trial segmenter options” which will requires to search among the “cleaned_image.jpg” to test options but the selected option should be transmitted to the segmenter.

In general: “the recall-options” should always comes with 4 buttons: “recall last used”; “load saved configurations”; “save configuration” and “reset to recommended options”

option “keep objects” should be “yes” by default

gromain, 2021-07-07:
A couple of typos:
- In the page name ("PlanktoSscope"): 
[image]

- In the first page ("RememVer"):
[image]

gromain, 2021-08-11:
For the following three items, fields were already there:
- modify as an (optional) field the “concentrated volume” ( add as datatip “i.e. volume obtained by the net cod end”)
- add an (optional) field for “net opening diameter” ( datatip ” diameter of mouth of the net”)
- add an (optional) field for “net mesh size” ( datatip ” mesh pore size”)

[image]

gromain, 2021-08-20:
>  add a button to recall previous options (number of images/pumped volume/delay between images)
Could this be a simple reload of the last used values on startup and a button to reset to default values (the other way around than what it is now)?

ethanjli, 2023-11-22:
This issue is too big (i.e. too many subtasks) and difficult to understand (at least for me) for it to be a good first issue, so I will remove the "good first issue" label from this issue.

My recommendation is that we should replace this issue with a bunch of smaller issues for specific groups of tasks, and then close this issue.