fairscope/PlanktoScope #92: Investigate alternatives for the median calculations for the flat field

gromain, 2021-07-05:
Some alternatives to a median calc exists, as for example [here](https://github.com/jiho/apeep/blob/master/scratch/process.py#L190) where someone uses an exponential decay. A mean could also be used. Tests needs to be done there.

ethanjli, 2023-11-22:
The work to characterize various algorithms and test them in the context of the PlanktoScope's segmenter will be rather involved, so I don't think this is a good first issue. So I will remove the "good first issue" label from this issue.

ethanjli, 2024-06-26:
Update: technical reference documentation of all the calculations made by the segmenter has now been published at https://docs-beta.planktoscope.community/reference/software/functionalities/segmentation/ (mostly added by #390). Currently @Oumayma-hy is working on investigating alternatives to some steps performed by our current segmenter implementation, but I believe she is not working on investigating alternatives to the median calculation for background subtraction.