fairscope/PlanktoScope #93: Use systemd to properly manage the python script (and raspimjpeg?)

gromain, 2021-07-05:
Right now, the python script is started/stopped/restarted by the flow in Node Red.

This is messy and creates a too tight coupling between Node Red and the python script. Border cases are not properly handled.

A transition to use a unit from systemd should be evaluated and implemented.

ethanjli, 2021-12-10:
I have a [working setup script](https://github.com/pez-globo/pufferfish-software/blob/develop/deploy/scripts/backend_service.sh) and [systemd unit](https://github.com/pez-globo/pufferfish-software/blob/develop/deploy/configs/pufferfish_backend.service) with Raspbian. Those files are configured for use with pyenv+poetry for Python dependency management, but a similar approach should work for whatever dependency management tooling you're using, if any. I'm happy to advise on this, or, with some onboarding guidance, to eventually create a PR.

gromain, 2021-12-12:
This is something that is quite simple indeed.

I'll have a look at your work on pufferfish, sounds like there is a look we can inspire from your work there.

closed 2023-08-31 by ethanjli

ethanjli, 2023-08-31:
Note: #187 doesn't run raspimjpeg as a systemd service - currently the Python script still starts and stops it. If we decide that it's important to run raspimjpeg as a systemd service, let's make a separate issue for that.

PR #187 (2023-08-31): Unify support for different HAT types into the same branch of the repo
This PR:
- Upgrades the provided Python backend to a version where the hardware controller is a separate Poetry project from the segmenter.
- Runs the Python backend as systemd services, rather than starting and stopping them from the Node-RED dashboard (fixes #93). The Node-RED dashboard now has two separate buttons for restarting the hardware controller and the segmenter; each button just issues a `systemctl restart` command for the respective systemd service.
- Removes the Node-RED dashboard's "Administration" page's emb [... cut]