fairscope/PlanktoScope #94: If mosquitto doesn't start properly, the python script crashes and doesn't recover on its own

gromain, 2021-07-06:
Rare scenarios can happen where Node Red starts before Mosquitto (for example when the network connection takes a while to configure). When the python script is then launched, it fails to connect to the Mosquitto server, with an ugly error.

This should be gracefully managed and the connection attempted again after a while.

Full error traceback:
```
2021-07-06 10:45:10.756 | ERROR    | planktoscope.mqtt:__init__:84 - An error has been caught in function '__init__', process 'StepperProcess-1' (630), thread 'MainThread' (3065821904):
Traceback (most recent call last):

  File "/home/pi/PlanktoScope/scripts/main.py", line 120, in <module>
    stepper_thread.start()
    │              └ <function BaseProcess.start at 0xb60dfb70>
    └ <StepperProcess(StepperProcess-1, started)>

  File "/usr/lib/python3.7/multiprocessing/process.py", line 112, in start
    self._popen = self._Popen(self)
    │    │        │    │      └ <StepperProcess(StepperProcess-1, started)>
    │    │        │    └ <staticmethod object at 0xb60e42b0>
    │    │        └ <StepperProcess(StepperProcess-1, started)>
    │    └ None
    └ <StepperProcess(StepperProcess-1, started)>
  File "/usr/lib/python3.7/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
           │                │                            └ <StepperProcess(StepperProcess-1, started)>
           │                └ <function DefaultContext.get_context at 0xb610cc90>
           └ <multiprocessing.context.DefaultContext object at 0xb60e4e90>
  File "/usr/lib/python3.7/multiprocessing/context.py", line 277, in _Popen
    return Popen(process_obj)
           │     └ <StepperProcess(StepperProcess-1, started)>
           └ <class 'multiprocessing.popen_fork.Popen'>
  File "/usr/lib/python3.7/multiprocessing/popen_fork.py", line 20, in __init__
    self._launch(process_obj)
    │    │       └ <StepperProcess(StepperProcess-1, started)>
    │    └ <function Popen._launch at 0xa69ae108>
    └ <multiprocessing.popen_fork.Popen object at 0xb5f26910>
  File "/usr/lib/python3.7/multiprocessing/popen_fork.py", line 74, in _launch
    code = process_obj._bootstrap()
           │           └ <function BaseProcess._bootstrap at 0xb60ea030>
           └ <StepperProcess(StepperProcess-1, started)>
  File "/usr/lib/python3.7/multiprocessing/process.py", line 297, in _bootstrap
    self.run()
    │    └ <function StepperProcess.run at 0xb53685d0>
    └ <StepperProcess(StepperProcess-1, started)>

  File "/home/pi/.node-red/projects/PlanktoScope/scripts/planktoscope/stepper.py", line 498, in run
    topic="actuator/#", name="actuator_client"

> File "/home/pi/.node-red/projects/PlanktoScope/scripts/planktoscope/mqtt.py", line 84, in __init__
    self.connect()
    │    └ <function MQTT_Client.connect at 0xb54635d0>
    └ <planktoscope.mqtt.MQTT_Client object at 0xb60e4990>

  File "/home/pi/.node-red/projects/PlanktoScope/scripts/planktoscope/mqtt.py", line 91, in connect
    self.client.connect(self.server, self.port, 60)
    │    │      │       │    │       │    └ 1883
    │    │      │       │    │       └ <planktoscope.mqtt.MQTT_Client object at 0xb60e4990>
    │    │      │       │    └ '127.0.0.1'
    │    │      │       └ <planktoscope.mqtt.MQTT_Client object at 0xb60e4990>
    │    │      └ <function Client.connect at 0xb545e300>
    │    └ <paho.mqtt.client.Client object at 0xa6a1f210>
    └ <planktoscope.mqtt.MQTT_Client object at 0xb60e4990>

  File "/home/pi/.local/lib/python3.7/site-packages/paho/mqtt/client.py", line 941, in connect
    return self.reconnect()
           │    └ <function Client.reconnect at 0xb545e420>
           └ <paho.mqtt.client.Client object at 0xa6a1f210>
  File "/home/pi/.local/lib/python3.7/site-packages/paho/mqtt/client.py", line 1075, in reconnect
    sock = self._create_socket_connection()
           │    └ <function Client._create_socket_connection at 0xb54630c0>
           └ <paho.mqtt.client.Client object at 0xa6a1f210>
  File "/home/pi/.local/lib/python3.7/site-packages/paho/mqtt/client.py", line 3546, in _create_socket_connection
    return socket.create_connection(addr, source_address=source, timeout=self._keepalive)
           │      │                 │                    │               │    └ 60
           │      │                 │                    │               └ <paho.mqtt.client.Client object at 0xa6a1f210>
           │      │                 │                    └ ('', 0)
           │      │                 └ ('127.0.0.1', 1883)
           │      └ <function create_connection at 0xb652e618>
           └ <module 'socket' from '/usr/lib/python3.7/socket.py'>
  File "/usr/lib/python3.7/socket.py", line 727, in create_connection
    raise err
          └ ConnectionRefusedError(111, 'Connection refused')
  File "/usr/lib/python3.7/socket.py", line 716, in create_connection
    sock.connect(sa)
    │    │       └ ('127.0.0.1', 1883)
    │    └ <method 'connect' of '_socket.socket' objects>
    └ <socket.socket [closed] fd=-1, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=6>

ConnectionRefusedError: [Errno 111] Connection refused
```

gromain, 2021-07-06:
Partially solved by f79cc62a608367533a2c15ce624d4b3f48de0c46

ethanjli, 2021-12-10:
Based on the architecture of the Python scripts, do you expect the solution to this issue will be more complicated than having a simple `while` loop? e.g. something like:
```
retrying_connection = False
while True:
    try:
        self.module_client = planktoscope.mqtt.MQTT_Client(...)
        break
    except ConnectionRefusedError:
        # print any logging statements
        time.sleep(retry_interval)
```
This is the [same approach](https://github.com/pez-globo/pufferfish-software/blob/ab5ac72f49ce3711f18544ce3059855b020671f6/backend/ventserver/io/trio/endpoints.py#L35) I used in another project, for starting a Python program and making it wait-and-retry establishing a pyserial connection, as well as wait-and-retry establishing a websocket connection to a client in a web browser.

Once the Python script is managed by systemd, I think it should be possible to configure the systemd unit to depend on the mosquitto service, so that systemd will never start the Python script before mosquitto finishes starting. But I think it would still be good for robustness and portability beyond Raspbian/systemd to make the Python script tolerant of not having the MQTT server at startup.

gromain, 2021-12-12:
Yes, exactly what I had in mind. And yes, very much linked to #104. Though 104 was there to try to figure also what should be happening with the script if the mosquitto server is not there (or if it dies). Should the script just shut down everything and clean up or should it keep on retrying, or try to force start the server by issuing a systemd call? Question is open for now.

Right now, the python script is started and managed by nodered (which itself is modified to depend on mosquitto). So unless the server dies, this should not be an issue.

ethanjli, 2024-04-11:
@melissadjadoun this is also related to what you were talking about with MQTT error-handling in the [2024-04-11 software meeting](https://docs.google.com/document/d/1Iba0_9yNFEDdp1Qy8IYlLP43Ga0DnDr3CdavWV0QGZY/edit#heading=h.e3f0mm3ng2sj)

chevreuill3000, 2026-11-09:
Closing: the controller services are now managed by systemd and restart automatically if MQTT is not available yet. The remaining MQTT connection error handling is tracked in #104.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-11-09 by chevreuill3000

chevreuill3000, 2026-11-09:
Closing: the controller services are now managed by systemd and restart automatically if MQTT is not available yet. The remaining MQTT connection error handling is tracked in #104.