fairscope/PlanktoScope #95: RTC does not update the local clock on startup

gromain, 2021-07-26:
On startup, time should be restored to the RTC value.

This is not currently the case.

ethanjli, 2021-12-10:
I have a working setup script for the MAX3231 RTC with Raspbian [in another project](https://github.com/pez-globo/pufferfish-software/blob/develop/deploy/scripts/rtc_setup.sh), and in order to make it work, we disabled the `fake-hwclock` systemd service and we also modified the `/lib/udev/hwclock-set` system script. I'm new to this project but am interested in helping out - is there a place where you have shell scripts to run for setting up the OS image, e.g. any script to copy files from `scripts/raspbian_configuration` to `/`? I'm happy to advise on this, or, with some onboarding guidance, to eventually create a PR.

gromain, 2021-12-10:
I haven't transfered the pi-gen repository to PlanktonPlanet yet. But it will happen soon.
If you can already send the information here, it would be really helpful! We disable `fake-hwclock` but what happens is that on startup the clock is not restored from the RTC on the HAT. I'll look for the guide I was using as a base to see what we changed and what we didn't.

ethanjli, 2021-12-10:
Based on what you've mentioned (disabling `fake-hwclock`) and what I can see (the `/boot/config` `dtoverlay` configuration), the other possible difference I can think of right now is the `hwclock-set` script. The version we used in the other project can be found [here](https://github.com/pez-globo/pufferfish-software/blob/develop/deploy/configs/hwclock-set), and it was created just by following the script modification instructions from [the Adafruit guide](https://learn.adafruit.com/adding-a-real-time-clock-to-raspberry-pi/set-rtc-time).

ethanjli, 2023-09-06:
In the [2023-08-31 planktoscope software taskforce meeting](https://docs.google.com/document/d/1fW8OfzK9r2wZVe9-b4BBmfvQma5U5Fneso1usAfge6k/edit#heading=h.xj07vi8xkjvc), Fabian reported: "I did have some PlanktoScopes which suddenly (magically) got the correct time - they were in the middle of a boat, without GPS. That unit was used a lot and was connected to various devices." It's possible that that PlanktoScope got internet access via some other device at some point.

One test to run for troubleshooting is to make the v2.5 PlanktoScope get time from the internet and test out the RTC with the PlanktoScope disconnected from the internet. Manu took (for another research cruise) the v2.5 PlanktoScope I was using to test the software, and he instructed Adam to give me a v2.5 PlanktoScope (which Adam had been using) after Adam returns to lab, so afterwards I will be able to run that test.

ethanjli, 2024-01-15:
I double-checked the HAT schematic, and apparently the RTC used there is the [PCF85363](https://www.nxp.com/docs/en/data-sheet/PCF85363A.pdf). On the other hand, the software is currently [configured](https://github.com/PlanktoScope/PlanktoScope/blob/software/v2023.9.0/software/distro/setup/planktoscope-app-env/platform-hardware/boot/config.txt.snippet) for the RV3028 RTC (which appears to be the module used in [older versions of the custom PlanktoScope HAT](https://planktoscope.readthedocs.io/en/latest/hat_hardware/#rtc-rv-3028-c7)). So the next step here is to change the boot config to enable kernel support for the correct RTC module, and see if anything changes.

ethanjli, 2025-01-23:
On PlanktoScopes without an RTC, it would be nice if we could leave in place the default functionality (which currently gets disabled) to persist the system time onto the filesystem. That way, even if the clock is out-of-date then at least the python backend log file timestampts will be monotonically increasing. That could also be a fix for https://github.com/PlanktoScope/PlanktoScope/pull/467 .

ethanjli, 2025-03-20:
I realized that actually the [fake-hwclock](https://manpages.debian.org/bookworm/fake-hwclock/fake-hwclock.8.en.html) functionality is not disabled - instead, I believe what's happening is that at shutdown `fakehwclock.service`'s shutdown procedure is running after changes to `/etc/fake-hwclock.data` get unlayered as a result of the stop procedure of `overlay-etc.service`. Probably the workaround will be to add another systemd service (not sure if adding it to pallet-standard would suffice, but that would be worth testing) which manually copies `/etc/fake-hwclock.data` to `/sysroot/etc/fake-hwclock.data` at shutdown.

ethanjli, 2025-03-20:
@sonnyp I'm tagging you as an assignee of this issue because you have easier and more consistent access to the PlanktoScope v2.5+ hardware than I do, and because this might be a good entry point for you to get familiar with our OS build scripts. If we're lucky, this issue might be fixable just by modifying `/boot/config.txt` to select the correct RTC driver. But there are some unknowns which might make this issue tricker to resolve, e.g.:
- whether different HAT versions have different RTC chips - and, if so, whether they can all be supported simultaneously and whether they all have corresponding kernel drivers
- whether there might be any weird interactions with clock-setting functionality (as supported in PlanktoScope OS v2024.0.0 via https://github.com/PlanktoScope/PlanktoScope/blob/software/v2024.0.0/software/distro/setup/planktoscope-app-env/gps/install.sh + https://github.com/PlanktoScope/device-pkgs/blob/edge/core/host/planktoscope/gpsd/overlays/etc/chrony/conf.d/20-pps-gpio.conf + https://github.com/PlanktoScope/device-pkgs/blob/edge/core/host/planktoscope/gpsd/overlays/etc/default/gpsd) with the GPS HAT which was included as part of the v2.1 PlanktoScope hardware's standard design

ethanjli, 2025-03-26:
One other note which I just remembered is that any change we make here should not break support for the RPi 5's built-in RTC module.

ethanjli, 2025-04-01:
Ooh, thanks for figuring that out! In that case this issue will just be naturally resolved (for new PlanktoScopes in the upcoming v3 hardware design, and for existing PlanktoScopes whose owners replace the RPi 4 with an RPi 5 including Raspberry Pi's [RTC battery](https://www.raspberrypi.com/products/rtc-battery/)) once we complete https://github.com/PlanktoScope/PlanktoScope/issues/322; based on your changes to the documentation in https://github.com/PlanktoScope/PlanktoScope/commit/8a931f50cda40716e482096a2e7d035edd90af1b, for other hardware configurations there is no RTC hardware for PlanktoScope OS to support officially.