fairscope/PlanktoScope #100: Insert guard for when hardware.json is empty

gromain, 2021-11-19:
Here: https://github.com/PlanktonPlanet/PlanktoScope/blob/2a0d0f8dfccf3a6254a5baddf614669edcc141d8/scripts/planktoscope/stepper.py#L147

There is a need to insert a try-Exception block for when the file is present but is empty.
This can also be done by checking the file size or length.

clatrellu, 2021-11-24:
Proposition pour cet issue
A changer dans imager/__init__.py et stepper.py

default_config = {
                    "stepper_reverse": false,
                    "microsteps": 32,
                    "focus_steps_per_mm": 40,
                    "pump_steps_per_ml": 507,
                    "focus_max_speed": 0.5,
                    "pump_max_speed": 30,
                    "stepper_type": "adafruit",
                    "wb_red_gain": 2,
                    "wb_blue_gain": 1.41,
                    "analog_gain": 1.0,
                    "digital_gain": 1.0
                    }
        configuration = default_config
        try:
            if os.path.getsize("/home/pi/PlanktoScope/hardware.json") > 0 :
                # load hardware.json
                with open("/home/pi/PlanktoScope/hardware.json", "r") as config_file:
                    configuration = json.load(config_file)
                    logger.debug(f"Hardware configuration loaded is {configuration}")
            else:
                logger.info("The hardware configuration file is empty, using defaults")
        except OSError:
            logger.error("The hardware configuration file is missing, using defaults")

ethanjli, 2024-03-26:
It seems that the imager module already includes a fix like what has been proposed here; that approach has been carried over to the new (picamera2-based) imager module introduced by https://github.com/PlanktoScope/device-backend/pull/19. #106 also attempted to introduce the same fix for the stepper module, but I closed it without merging because it was a stale PR and there were too many changes to different parts of the software. As I incrementally rewrite the Python hardware controller module-by-module, I will try to ensure that we have adequate error-handling for the `hardware.json` file in every module.

ethanjli, 2024-06-08:
@Oumayma-hy FYI

chevreuill3000, 2026-11-11:
Closing as a duplicate of #552. The guard asked for here (a `hardware.json` that exists but is empty or invalid) is the cause identified in #552, and the work that remains, writing the file atomically and letting every controller start with a clear error, is described there. Please reopen if a separate issue is still useful.

closed 2026-11-11 by chevreuill3000