fairscope/PlanktoScope #104: MQTT: catch ConnectionRefusedError

gromain, 2021-11-23:
At https://github.com/PlanktonPlanet/PlanktoScope/blob/5b2bb0b9cdf81bd33fb8ca8d3d08f8feae546ea9/scripts/planktoscope/mqtt.py#L88
And decide what to do when it happens.

ethanjli, 2023-11-22:
In my opinion, the MQTT client should continuously attempt to retry connecting to the MQTT broker (mosquitto) every few seconds until the connection succeeds.

However, this kind of retry loop should not block any behaviors which need to occur regardless of whether the MQTT broker is connected - for example, fan control to regulate CPU temperature.

ethanjli, 2024-04-11:
@melissadjadoun this is the issue I mentioned in the [2024-04-11 software meeting](https://docs.google.com/document/d/1Iba0_9yNFEDdp1Qy8IYlLP43Ga0DnDr3CdavWV0QGZY/edit#heading=h.e3f0mm3ng2sj) that's related to what you were talking about