fairscope/PlanktoScope #108: Create Jupyter Sandbox

gromain, 2021-11-25:
Would be nice to integrate a Jupyter notebook to be able to play with the data generated.

They exists in a web available format already, so it could be as simple as integrating a template.

ethanjli, 2023-11-22:
Normally, it should be straightforward to add an app like JupyterLab as a Docker container (via https://github.com/PlanktoScope/device-pkgs and https://github.com/PlanktoScope/pallet-standard ). However, it appears that Jupyter's [official Docker images](https://quay.io/organization/jupyter) aren't built for armv7, which is what we're stuck on until we migrate to picamera2 (#79). In the meantime, we'd have to either identify Jupyter Docker container images built for armv7, or make our own builds for armv7. https://hub.docker.com/r/andresvidal/jupyter-armv7l exists, but it is unmaintained (it only ever had one release, 4 years ago) - so we can't use it, but it might be a useful starting point for us to maintain our own armv7 builds of Jupyter Docker container images.

My guess is the need for architecture-specific builds comes from the various numerical package dependencies of numpy/scipy/etc.

ethanjli, 2024-01-16:
Update: I discovered that Go binaries built for aarch64 work on our 32-bit version of RPi OS, so maybe we don't actually need to make an armv7 build. The next step here is to just try seeing if the official docker container images will work.

ethanjli, 2024-05-25:
Once we merge #416, an incremental strategy for completing this issue might look as follows:

- Make a JupyterLab sandbox an opt-in (i.e. disabled-by-default) Forklift package, so we can try out JupyterLab in isolation
- Add integration with `/home/pi/data` so we can use JupyterLab to manipulate PlanktoScope datasets
- Provide an MQTT client library which the JupyterLab sandbox can use to script/automate operation of the Python segmenter
- Enable use of the Python segmenter's functionalities as a library in the JupyterLab sandbox, without requiring an MQTT broker or a separate Python segmenter process
- Provide an MQTT client library which the JupyterLab sandbox can use to script/automate operation of the Python hardware controller

closed 2026-11-09 by chevreuill3000