fairscope/PlanktoScope #109: Segmenter crashing after calculating flat

babo989, 2021-11-30:
**Describe the bug**
As soon as the segmentation is started, the scope calculates flat field, then moves on to maybe 1 or 2 images. After this-the Python crashes. Restarting Python will fix the functionality, but not the segmentation. 

**To Reproduce**
Steps to reproduce the behavior:
1. Complete an acquisition project (50 images). Navigate to 'segmentation' and select the newest folder. Begin segmentation.
2. Wait until the segmentation starts, it usually hangs up after 2 or 3 images. 

**Expected behavior**
Ideally, segmentation should finish with the ecotaxa archive completed.

**Screenshots**
2021-07-01 18:23:12.398 | DEBUG | planktoscope.segmenter:_pipe:635 - The debug objects path is /home/pi/data/clean/20211107/Peggy's_Bench__1/Peggy's_Bench__1_32/18_14_49_742246
2021-07-01 18:23:12.405 | INFO | planktoscope.segmenter:_pipe:642 - Starting work on 18_14_49_742246, image 1/20
2021-07-01 18:23:12.408 | INFO | planktoscope.segmenter:_open_and_apply_flat:189 - Opening images
2021-07-01 18:23:13.005 | SUCCESS | planktoscope.segmenter:_open_and_apply_flat:198 - Opening images
2021-07-01 18:23:13.006 | INFO | planktoscope.segmenter:_open_and_apply_flat:200 - Flat calc
2021-07-01 18:23:20.437 | ERROR | __main__:<module>:161 - The segmenter process died unexpectedly! Oh no!
2021-07-01 18:23:20.487 | INFO | __main__:<module>:165 - Shutting down the shop
2021-07-01 18:23:20.520 | INFO | planktoscope.imager:run:713 - Shutting down the imager process
2021-07-01 18:23:20.521 | INFO | planktoscope.stepper:run:526 - Shutting down the stepper process
2021-07-01 18:23:20.523 | DEBUG | planktoscope.imager:run:715 - Stopping the raspimjpeg process
2021-07-01 18:23:20.525 | DEBUG | planktoscope.imager.raspimjpeg:close:484 - Killing raspimjpeg in a nice way
2021-07-01 18:23:20.538 | INFO | planktoscope.mqtt:shutdown:165 - Shutting down mqtt client actuator_client

**Additional context**
This is on the newest version of the scope, with the most up to date image release.

gromain, 2021-11-30:
Thanks for the report! We've had a lot of issues with the beta version of the segmenter.

Can you share the sample you're trying to segment? Was it overcrowded?

ethanjli, 2023-11-22:
Since we have received no follow-up information in the past 2 issues about the specific conditions in which this issue occurred, and thus we have no hope of reliably reproducing this problem, I will close this issue. If anyone encounters the same issue in the future, please attach the log from the segmenter as well as the raw dataset you were trying to process.

closed 2023-11-22 by ethanjli