fairscope/PlanktoScope #110: Create a hardware settings module registering the machine configuration

gromain, 2021-12-01:
In the Hardware settings tab:
- [ ] Add an input text to enter the Production Serial Number (Written below the machine from now (ex:V2.5D26024))

In the Home tab:
- [ ] Add a group named Hardware Settings showing the current config : 
  - [ ] Production Serial Number
  - [ ] Machine version.
  - [ ] Camera Type
  - [ ] Stepper controller type.

ethanjli, 2023-11-10:
@LaurentPV I don't know what the purpose of this feature request is (or what would be the use of the data entered in the requested input fields), but since you're working on things related to hardware metadata, this issue might be of interest to you.

LaurentPV, 2023-11-13:
@ethanjli  Ok you can assign it to me.

closed 2026-11-09 by chevreuill3000

chevreuill3000, 2026-11-09:
Closing as done: the hardware version is now selected on the first-time setup page and the machine configuration is stored in `hardware.json`.