fairscope/PlanktoScope #112: Track usage statistics (locally)

gromain, 2021-12-01:
It would be nice to know how many acquisitions and segmentations were done and how many objects were found.
We need to create a way to track those statistics.

In the Home tab:
- [ ] Add a group named Statistics containing  : 
  - [ ] Number of acquisition performed
  - [ ] Number of segmentation performed
  - [ ] Total number of objects detected

ethanjli, 2023-11-10:
Implementing this will probably require a database (which can be an SQLite database or even as simple as a JSON file, depending on the complexity of statistics we want to track) to persist those statistics across dataset file transfers.

ethanjli, 2023-11-22:
Since choosing a persistence mechanism needs to be a careful design decision, I don't consider it a good first issue for newcomers to go ahead and implement; rather, discussion (and perhaps a [design document](https://github.com/PlanktoScope/proposals/)) will be needed. Thus, I am removing the "good first issue" label from this issue.

chevreuill3000, 2026-11-09:
Closing as a duplicate of #1077 (telemetry on the PlanktoScope).

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-11-09 by chevreuill3000

chevreuill3000, 2026-11-09:
Closing as a duplicate of #1077 (telemetry on the PlanktoScope).