fairscope/PlanktoScope #118: Add a button to activate/deactivate the wifi module

gromain, 2021-12-03:
Right now, a change to `/boot/config.txt` is needed to deactivate the wireless module.

It would be nice to have a button that would do so on the next boot of the machine.

Sidahmed-ben, 2022-03-21:
Hello @gromain  , I'm interested by the issue if you agree i can find a solution 😉

ethanjli, 2023-11-22:
I'm not sure what the motivation for this issue is. I can think of two possible reasons we might want to disable the wireless module:

1. To save battery, by turning off the wifi circuitry.
2. To prevent the PlanktoScope from creating a wifi hotspot network, even when the PlanktoScope is not connected to an existing wifi network (e.g. for security reasons, to prevent anyone who walks by from being able to control the PlanktoScope).

If reason 1 is important, I agree that we will need to change `/boot/config.txt`. However, we'll need to think through how to ensure that the GUI is always consistent with the current state of the wifi module and the state of the wifi module which will be applied upon the next boot. I think there will be some complexity in this.

If reason 1 is not that important, we should just leave the wifi module on but disable the PlanktoScope's autohotspot functionality. However, this will require an understanding of how we have implemented the autohotspot service.

In either case, I think this issue will take some discussion and thoughtful design, as well as some knowledge of how we are managing the behavior of the Raspberry Pi's network interfaces (what we are doing is not yet documented). So I don't think this issue is a good first issue for a newcomer to go ahead and implement, and I will remove the "good first issue" label from this issue.

ethanjli, 2025-03-20:
As of PlanktoScope OS v2024.0.0, we now have a guide for advanced users to disable the Wi-Fi hotspot at https://docs.planktoscope.community/operation/networking/#disable-the-wi-fi-hotspot .

Because a user without immediate access to an Ethernet cable (+ adapter, if needed) or SD card reader who disables the Wi-Fi hotspot would lose access to their PlanktoScope until they can locate either an Ethernet cable or an SD card reader, I think it is too risky to give users a GUI button for disabling the Wi-Fi module; instead, I think it's better for now to hide this process behind our relatively-scary guide with commands to run in the Cockpit terminal and with warnings about making hard-to-reverse changes. So I'm going to close this issue as resolved now. If we later change our minds and decide that we want a GUI button with appropriate GUI safeguards, we can reopen this issue.

closed 2025-03-20 by ethanjli