fairscope/PlanktoScope #132: Distribute third-party applications as Docker Stacks in a Github repository

ethanjli, 2023-04-27:
## Motivation

Currently, our PlanktoScope software distro uses apt-get to install all third-party applications (e.g. Mosquitto, Nginx, Node-RED). Upgrading such applications (e.g. to receive bugfixes or security patches) can be done by running `apt-get install <package_name>=<version>` in a migration script, though we would need to be careful that any dependencies installed as a result are of reproducible, consistent versions. More complicated, however, is that changing the configuration files for such applications would require making migration shell scripts and keeping track of how the state of the OS evolves - and that's not to mention the potential consequences of user actions which may make the OS diverge into a state inconsistent with what the migration scripts expect (e.g. if a user runs `sudo apt-get update; sudo apt-get upgrade`). This is a class of problems resulting from treating the OS as [mutable infrastructure](https://www.hashicorp.com/resources/what-is-mutable-vs-immutable-infrastructure); we avoid it by asking users to just re-flash their SD cards with a fresh image, which treats the OS as immutable infrastructure to avoid the problems of mutability. However, re-flashing SD cards requires internet bandwidth (to download the SD card), time (to flash the SD card), planning (to prevent loss of data), and potentially reconfiguration (to change the software's hardware settings again to match the actual hardware configuration).

By distributing applications as Docker Stacks which specify Docker container images together with config files, we reduce the mutability for applications (which may need to update frequently, in contrast to low-level OS packages which may only need to update once every few years) without having to re-flash SD cards; we can just take down the old version of the application and bring up the new version in total isolation - and any application data would typically persist between upgrades. This greatly reduces the risk to users that installing a software upgrade through a migration script could break their OS, but still without the need to re-flash their SD card.

Also, using Docker for third-party applications makes it feasible for developers to develop and test against those applications on our own development machines, in a reproducible way.

## Goals

- To upgrade/downgrade third-party applications, the user will run a one-line command to download and run a simple migration script from a Github repository. That migration script will perform actions to download and apply Docker Stack files.

## Steps

- [x] Make a Github repo which will act as our repository of Docker Stack collections for the PlanktoScope software distro, with a root-level manifest and sub-directories for different flavors and sub-sub-directories for different versions: https://github.com/PlanktoScope/pallets contains all software packages (though other people can fork it for custom flavors), and https://github.com/PlanktoScope/pallets-env is one flavor (a set of configurations of packages to deploy); versioning will be done using Git tags
- [x] Include a Docker Stack for Portainer as a GUI for advanced users to inspect & monitor the deployed applications: https://github.com/PlanktoScope/pallets/blob/b35f9cc9837f11092391263b8ee1682c22aa04f4/core/portainer/pallet-package.yml
- [x] Include a Docker Stack for Caddy as a reverse-proxy: https://github.com/PlanktoScope/pallets/blob/47866882897b922c0685c0ac2a3823a978a1f8c5/core/caddy-ingress/pallet-package.yml
- [x] Include a Docker Stack for filebrowser as the file management interface for datasets: https://github.com/PlanktoScope/pallets/blob/47866882897b922c0685c0ac2a3823a978a1f8c5/core/filebrowser-datasets/pallet-package.yml
- [x] Make a declarative specification of Docker Stack collections: https://github.com/PlanktoScope/pallets-env
- [x] Make a CLI tool to apply Docker Stack collections for upgrades/downgrades: https://github.com/PlanktoScope/forklift
- [ ] Set up Docker (with Swarm Mode) as part of the SD card image; we may need initialization of swarm mode to happen after the SD card is flashed, to avoid sharing private keys
- [ ] Integrate our packaging system into the process for building SD card images
- [ ] Reconfigure the Node-RED dashboard for reverse-proxying from Caddy
- [ ] Enable users to pin their software deployment to "stable", "beta", and "edge" release channels (add "stable" and "beta" branches)

## Unresolved Questions

- Installing Cockpit onto the v2.3 PlanktoScope SD card image seems to break internet connectivity; uninstalling it restores DNS lookup functionality but not the ability to connect to things on the internet - why is this breakage happening?
- Installing Docker and setting up Docker Swarm onto the v2.3 PlanktoScope SD card image seems to interact with dhcpcd in a way that breaks the autohotspot service (the autohotspot service needs to be restarted in order for the PlanktoScope to get internet access over wifi) - we need to fix this in order to include Docker in the next software release

## Implementation History

- forklift: everything from the creation of the repository up to https://github.com/PlanktoScope/forklift/releases/tag/v0.1.2
- pallets: everything from the creation of the repository up to https://github.com/PlanktoScope/pallets/commit/b35f9cc9837f11092391263b8ee1682c22aa04f4
- pallets-env: everything from the creation of the repository up to https://github.com/PlanktoScope/pallets-env/commit/18e66cac286211404c593833e813d812cdebd6c4
- Video demo: https://www.youtube.com/watch?v=FgtC38aecaE

gromain, 2023-05-10:
The only limitation with docker will be that low-level access to hardware (for the pscope stack itself) will be blocked if we encapsulate node-red and/or the python scripts.

Otherwise, the rest makes sense.

ethanjli, 2023-05-10:
Yes, those are also concerns I had - for these reasons, I listed them as separate tasks in #133. I've added a note in this PR to clarify this point.

I assume that if we run the python scripts in a Docker container in [privileged mode](https://docs.docker.com/engine/reference/commandline/run/#privileged). I would really like to avoid using privileged mode if we can avoid it (because Docker won't create services in privileged mode), in which case hopefully I can just find a set of bind mounts and [capabilities](https://docs.docker.com/engine/reference/run/#runtime-privilege-and-linux-capabilities) which would be sufficient to make the python scripts work. This will depend on how libraries like RPi.GPIO, Adafruit-PlatformDetect, and Adafruit-Blinka interact with the OS.

Because the current Node-RED frontend executes various shell commands, it will be more problematic to run in a Docker container. In the long-term I would like to remove all shell commands from the frontend anyways (for clearer separation of responsibilities, and so that we could deploy the frontend on other computers). But that will take some work + time, so in the short-term I may need to try something like https://stackoverflow.com/questions/32163955/how-to-run-shell-script-on-host-from-docker-container which acts as an escape-hatch for the security sandboxing of Docker containers - so for the Node-RED frontend we'd only be using Docker as a software packaging/distribution/deployment format.

ethanjli, 2023-06-01:
@gromain Note: I will need to complete the "Make the Node-RED dashboard embed the reverse-proxied MJPEG streams" task in approx. 6 days - if #145 isn't ready to merge before then, I will be introducing a merge conflict in the flows.json file, and I will add a comment on the relevant PR describing the changes I made (and tagging you) so that you can manually replay those changes on PR #145 so that you can resolve the merge conflict by starting with your changes and just replaying my changes on top.

ethanjli, 2023-06-01:
@gromain Just an update, since I will be serving as a last-minute replacement for someone on my lab who very recently dropped out of a 45-day fieldwork trip to the Arctic Ocean (so I will have very limited internet usage quotas from mid-June to the end of July, and I definitely won't be able to run the setup scripts or download/upload SD card images). So I am moving the planned date of our next software release from July 2023 to September 2023 as a result, and I will avoid making any changes to the Node-RED flows.json file until after July 31 - so you won't need to worry about any Node-RED-related merge conflicts before then!

ethanjli, 2023-09-14:
This issue is complete (I've removed the final task of making a new "stable" branch, since there's nothing new for doing that compared to making a "beta" branch).

closed 2023-09-14 by ethanjli

ethanjli, 2023-12-19:
Starting with Raspberry Pi OS 12 (bookworm), we may be able to take advantage of https://systemd.io/PORTABLE_SERVICES/ and/or https://www.freedesktop.org/software/systemd/man/latest/systemd-sysext.html for things like the autohotspot service and things which need to exist before Docker is brought online.