fairscope/PlanktoScope #133: Make it safe, simple, and easy for users to apply software updates

ethanjli, 2023-04-27:
This is a tracking issue for our epic to improve our project's software update process in order to reduce the barriers for users to update their PlanktoScope software deployments.

## Motivation

Currently, safely updating the software on a PlanktoScope requires flashing a new SD card and swapping it out with the old SD card (as described in #88, the previous Git-based system was too finicky to be usable); in order to make it easy to downgrade the software, the old SD card (with the old OS image) must be kept around indefinitely instead of being reassigned for other purposes. Re-flashing SD cards requires internet bandwidth (to download the new SD card image), time (to flash the SD card), planning (to prevent loss of data), potentially spare SD cards (if the device owner wants to keep old SD cards for easy downgrades), and potentially reconfiguration (to change the software's hardware settings again to match the actual hardware configuration).

## Goals

- Reduce the number of software updates requiring an SD card re-flash (i.e. an irreversible upgrade) to a frequency of once every (KPI) four years.
- For reversible upgrades:
  - Reduce the total size of files which must be downloaded to less than (KPI) 100 MB.
  - Enable upgrading/downgrading the software to the next/previous release with (KPI) one button-press and a confirmation dialog.
  - Enable automatic background downloading of software artifacts for software upgrades.
  - Provide a mechanism to prevent loss of data and settings through a software update.
  - Make upgrades atomic.
- For irreversible upgrades:
  - Provide a mechanism to irreversibly upgrade the OS to a specific set of package versions.
  - Make upgrades atomic.

## Steps

(these steps will be converted into tracking issues for stories)

- [ ] Distribute third-party applications as Docker Stacks in a Github repository https://github.com/PlanktoScope/PlanktoScope/issues/132
- [ ] Make a landing page at http://planktoscope.local for discovery and easy access of all installed browser applications
- [ ] Make a GUI for easily upgrading/downgrading software
- [ ] Containerize the Node-RED frontend and run it in a Docker Stack
- [ ] Containerize the Python backend and run it in a Docker Stack (with separate packages or separate feature flags for different hardware configurations if necessary)
- [ ] Provide migration scripts which users can download to irreversibly upgrade their base OS (i.e. anything not in Docker Stacks).
- [ ] Enable detection & repair of drift from the declarative specification of package deployments

## Unresolved Questions

ethanjli, 2023-08-26:
Since the Node-RED dashboard runs host commands, containerizing it will be a bit tricky. https://stackoverflow.com/a/63140387 and https://security.stackexchange.com/a/218379 provide some hints about how we might do it, starting with a privileged container. We could then gradually lock down the frontend.

ethanjli, 2023-09-20:
For the base OS, it might be worth investigating [Yocto](https://medium.com/nerd-for-tech/build-your-own-linux-image-for-the-raspberry-pi-f61adb799652), either to use or as inspiration.

ethanjli, 2023-11-07:
It might be worth investigating whether we can deliver PlanktoScope OS-level binaries and scripts via a container image for [Distrobox](https://github.com/89luca89/distrobox) - refer to https://distrobox.it/usage/distrobox-export/ . We would need to evaluate whether we can integrate Distrobox with Forklift.

We might also be able to use chezmoi to manage config files (especially in the home directory) - we would need to evaluate whether we can integrate chezmoi with Forklift.

ethanjli, 2023-11-08:
https://www.youtube.com/watch?v=gSePCve0yZI discusses making configuration files easier to upgrade.

ethanjli, 2023-11-08:
https://www.youtube.com/watch?v=XTy3scX6rF4 discusses the use of filesystem images (which from my understanding are not container images) to manage config files like in `/etc`.

ethanjli, 2024-01-12:
https://github.com/ublue-os/sysext shows an example of how to use systemd-sysext, for anything which we'd need to distribute outside `/etc` which we can't just distribute using forklift. We could probably distribute shell scripts as part of forklift packages, and run them from the cached packages (by making a forklift subcommand to return the path of files in the package used by a particular app deployment, and using that path to invoke the script).

ethanjli, 2024-02-01:
https://github.com/mendersoftware/mender uses A/B partitions for atomic OTA updates, and it provides some kind of support to package up Debian-based images (so we should be able to stay on Raspberry Pi OS!)!

ethanjli, 2024-02-01:
https://www.boulderes.com/resource-library/building-raspberry-pi-disk-images-with-docker-a-case-study-in-software-automation provides a guide for building a Docker container image from a Raspbian base image, and for exporting it to a Raspberry Pi SD card image (!!). However, the Raspbian base images are ancient and there are no official Raspberry Pi OS base images on Docker Hub (though https://forums.raspberrypi.com/viewtopic.php?t=359999 has a post with [a script](https://gist.github.com/satmandu/0c6f560549d226cf352d48ca8c168315) to generate such base images).

ethanjli, 2024-04-17:
https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#fail-safe-os-updates-tryboot and https://www.raspberrypi.com/documentation/computers/config_txt.html#example-update-flow-for-ab-booting describe an officially-supported A/B system boot mechanism for the Raspberry Pi!

closed 2026-11-09 by chevreuill3000

chevreuill3000, 2026-11-09:
Closing as done: over-the-air updates are now available (RAUC A/B updates and the Update page of the dashboard), see #923.