fairscope/PlanktoScope #136: Improve & automate the process of building release artifacts

ethanjli, 2023-04-27:
This is a tracking issue for our epic to automate our project's software release process in order to reduce the work needed for each release.

## Motivation

Currently, there is an entirely manual process of creating a "golden image" for the RPi SD card, which is how we release the software distro; and currently, only Romain has been making the golden images. Many steps of the process appears to be documented at https://planktoscope.readthedocs.io/en/latest/expert_setup/, though others remain undocumented (for example, timezone and keyboard settings). If we can make the process mostly automated, then it will be easier for this project to release software updates even as the set of active software maintainers/contributors changes.

## Goals

- Make shell scripts to automate over (KPI) 90% of the steps necessary to create a golden image.
- Make a unified, simple-to-read, simple-to-understand, and simple-to-apply step-by-step guide/checklist of all manual actions necessary to create the golden image.

## Steps

(these steps will be converted into tracking issues for stories)

- [ ] https://github.com/PlanktoScope/PlanktoScope/issues/135
- [ ] Make a shell script to run on a separate computer for reproducibly making a shrunken SD card image from the SD card
- [ ] Make a Github Actions CI pipeline for building the SD card image, maybe using [pi-gen-action](https://github.com/usimd/pi-gen-action) (#42)

## Unresolved Questions

gromain, 2023-05-10:
Hi @ethanjli,

Sorry I didn't followed properly the developments as I was not tasked with this anymore.

For the golden image creation, there is already a pi-gen repository for the Planktoscope at https://github.com/PlanktoScope/pi-gen

Only restriction is that is has to be run on a Raspberry pi since the pip installation of some packages is somewhat finicky (and detects the architecture when building some packages).

I'm bringing this up to date to create the images for the Atlanteco and Biocean5D deployments, over at https://github.com/oceantrotter/pscope-pi-gen/tree/arm64 .

I'll create a PR once this is stabilised.

ethanjli, 2023-05-10:
Hi, @gromain - thanks for the update about what you've been doing! I think it will be more maintainable if we find a way to use pi-gen as a dependency or a base layer image rather than forking it for our own purposes, making a ton of modifications on our fork, and then pulling in changes from upstream as the upstream updates. For example, if I can get https://github.com/usimd/pi-gen-action to work with the setup scripts I've made in this PR (which would be my next PR after this one), then I think we could just run our own setup scripts as a custom user stage with pi-gen. Additionally, if we have a setup script which can be run on its own, then we can provide that for users to run on other base images of their own choosing - for example, a Raspberry Pi OS image with the graphical desktop. Finally, having our own independent setup script would avoid increasing the amount of tightly-coupled dependencies from our software on the Raspberry Pi OS and would provide a future path to setting up our software distribution on other platforms/targets if needed  - which would be important if, for example, we may need to make the software distribution deployable on more powerful compute hardware such as the Nvidia Jetson Nano or a similar platform for faster on-board segmentation.

Regarding your observations about issues with pip installation of packages, architecture detection, and the need to build the image on a Raspberry Pi - did you have these problems even when running pi-gen inside a virtualized 32-bit armhf guest OS in a virtual machine, e.g. in QEMU/VirtualBox/etc.? e.g. using https://github.com/marketplace/actions/arm-runner . And do you remember what specific issues with pip installation you encountered?

ethanjli, 2023-05-10:
Hi, @gromain - thanks for the update about what you've been doing! I think it will be more maintainable if we find a way to use pi-gen as a dependency or a base layer image rather than forking it for our own purposes, making a ton of modifications on our fork, and then pulling in changes from upstream as the upstream updates. For example, if I can get https://github.com/usimd/pi-gen-action to work with the setup scripts I've made #135, then I think we could just run our own setup scripts as a custom user stage with pi-gen. Additionally, if we have a setup script which can be run on its own, then we can provide that for users to run on other base images of their own choosing - for example, a Raspberry Pi OS image with the graphical desktop. Finally, having our own independent setup script would avoid increasing the amount of tightly-coupled dependencies from our software on the Raspberry Pi OS and would provide a future path to setting up our software distribution on other platforms/targets if needed  - which would be important if, for example, we may need to make the software distribution deployable on more powerful compute hardware such as the Nvidia Jetson Nano or a similar platform for faster on-board segmentation.

Regarding your observations about issues with pip installation of packages, architecture detection, and the need to build the image on a Raspberry Pi - did you have these problems even when running pi-gen inside a virtualized 32-bit armhf guest OS in a virtual machine, e.g. in QEMU/VirtualBox/etc.? e.g. using https://github.com/marketplace/actions/arm-runner . And do you remember what specific issues with pip installation you encountered?

ethanjli, 2023-09-14:
Usage report, now that I've gone through the prerelease process for v2023.9.0-beta.0 and v2023.9.0-beta.1:

Currently the most annoying part of building an SD card image is how long it takes (~30 minutes, much of which is installing dependencies for the Python backend and installing Node-RED for the Node-RED dashboard), and that I have to take various manual actions (flash the SD card with a fresh Raspberry Pi OS image, insert the SD card into the Raspberry Pi, manually run a set of commands). I think the order of priorities should be:
1. Containerize the Python backend and Node-RED dashboard so that I don't have to keep reinstalling the same dependencies over and over when I need to re-run the SD card setup scripts in order to iterate on some unrelated functionality.
2. Make the SD card setup process work in GitHub Actions CI.

This is because an extremely slow CI pipeline for building SD card images will still be really annoying for iterative development and testing.

ethanjli, 2023-11-30:
https://github.com/gitbls/sdm might also be useful as part of the build process.

ethanjli, 2024-01-06:
https://github.com/aniongithub/pi-bootstrap/ is also worth investigating for whether we might be able to use [pimod](https://jonashoechst.de/assets/papers/hoechst2020pimod.pdf) as a potential alternative to pi-gen-action

ethanjli, 2024-01-09:
After reviewing the design of pimod as described in the [pimod paper](https://jonashoechst.de/assets/papers/hoechst2020pimod.pdf) and reviewing https://github.com/aniongithub/pi-bootstrap/, I've decided that the first approach I will try will be pimod. It has the following advantages over pi-gen-action:

- Its "Pifile" spec is designed to be familiar to "Dockerfile"/"Containerfile" specs in syntax and semantics
- It can run in GitHub Actions (including integration with env vars provided via GitHub Actions), but it can also be used without GitHub Actions
- It will be easy for us to use it together with our existing setup scripts, allowing for incremental migration if we want to move things more into "Pifile" spec files, or (as I would prefer) allowing us to keep everything in our existing bash setup scripts so that we can also manually run setup without relying on pimod.

ethanjli, 2024-02-01:
As an alternative, https://github.com/PlanktoScope/PlanktoScope/issues/133#issuecomment-1922417905 suggests a way to build the OS into OCI images and SD card images without using pimod.

chevreuill3000, 2026-11-09:
Closing as a duplicate of #730 (build images automatically).

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-11-09 by chevreuill3000

chevreuill3000, 2026-11-09:
Closing as a duplicate of #730 (build images automatically).