fairscope/PlanktoScope #166: When an invalid value in the white balance is set, the imager process is crashing after reboot.

glx314, 2023-07-13:
**Describe the bug**
When an invalid value in the white balance is set, the imager process is crashing. This prevents the Planktoscope to set up correctly during the boot process.

**To Reproduce**
Steps to reproduce the behavior:
1. Go in the UI / optical configuration
2. Set the white balance to an invalid value (for us it was (8,1))
3. Reboot
4. Go in the UI / optical configuration
5. The video stream is not started.

**Expected behavior**
Not allowed to set invalid value + imager not crashing when invalid values are used (reset them to some defaults ?)

ethanjli, 2023-07-18:
## Reproduction

I have also encountered this problem in the past - the only way I could restore the PlanktoScope software to a working state was to repair or delete the hardware.json file.

## Analysis

I believe the root cause of this problem is that the software does not reject invalid white balance values, but instead writes them to the hardware.json file; then, after the next boot of the PlanktoScope, it tries to use those files to update the camera's white balance and then fails.

## Impact

I experienced this bug while running a PlanktoScope training workshop where I let participants play around with the Node-RED dashboard on their own; I didn't realize there was a problem until the subsequent day, when I found that the PlanktoScope had stopped working even though it had worked on the previous day. I was not able to troubleshoot the cause of this problem until after the workshop.

I also received a report of the same problem from someone I gave a PlanktoScope to afterwards, even though I had previously told him about my experience with the white balance values. He was able to figure out the cause of the problem on his own - he reported that he might have accidentally set one of the white balance values to `-1`.

Based on the fact that three people have independently encountered the same problem in the past six months, I am assigning this issue to to high priority, so I will try to include a fix in the next software release (scheduled for September).

## Proposed Fix

I believe the way to fix this bug will be to add input validation to the Python backend to ensure that it rejects invalid white balance values, and to add input validation to the Node-RED dashboard to prevent the user from inputting invalid white balance values.

ethanjli, 2023-08-29:
On the frontend side, it looks like the Node-RED dashboard uses text input nodes in "number" mode for white balance values; such nodes don't have a way to set max & min limits. The alternatives are:
1. (preferred, if feasible) Replace the text input nodes with number input nodes
2. (would have usability issues) Replace the text input nodes with slider nodes. This is not preferable because we often want to reproduce white balance values exactly, and a slider makes this difficult.
3. Keep the text input nodes, but clamp values within a 1.0 to 8.0 range. If possible, display a notification when a value is clamped. This is not preferable because it would be easier to just prevent the user from entering an invalid value.

PR #380 (2024-03-26): Switch from raspimjpeg to picamera2 for image acquisition
This PR fixes #79, fixes #206, fixes #212, fixes #217, and fixes #166 by bumping the Python hardware controller to incorporate the changes introduced in https://github.com/PlanktoScope/device-backend/pull/19 .
closed 2023-09-02 by ethanjli