fairscope/PlanktoScope #204: Automatic calculation of manual white balance values

ethanjli, 2023-09-05:
**Is your feature request related to a problem? Please describe.**
Currently, the manual white balance values must be iteratively and manually tuned based on visual appearance. This is not very rigorous, and hard to ensure good calibration between machines.

**Describe the solution you'd like**
A more rigorous way to tune white balance, used by Fabien, is to capture an image with no flowcell/ibidi chamber, and then use software to measure the mean RGB value of the image. Then that value can be used to calculate the white balance gains needed to achieve some desired mean RGB value. Fabien requests that the software provide an interface to automate the adjustment of the manual white balance gains (e.g. after the user presses a single button).

**Describe alternatives you've considered**
Another option which would be somewhat less automated and less integrated into the software, but which could be an intermediate prototype provided as an experimental feature in one release and then integrated more fully in a subsequent release, could be to provide a standalone tool where the user pastes in an image with no flowcell/ibidi chamber, and then the tool proposes a pair of manual white balance gains to set. Then the user can paste in an image acquired with those gains, and either the software tells the user that the gains are good (if only one iteration of trial-and-error for gain adjustment is required) or it provides them the next pair of gains to try (if white balance adjustment must be an iterative process).

ethanjli, 2023-09-06:
This feature request is not in the critical path for my thesis work, so I will not prioritize implementing it during my thesis. However, I would be happy to mentor someone else who wants to implement this feature.

I would suggest first making a small standalone web app which takes an image upload (or ideally a user pasting an image from their clipboard, since it's easy to copy the current frame of the camera preview stream) and then calculates and displays the white balance gains to recommend to the user. We can then distribute this as an experimental browser app in the PlanktoScope software distro, for users to test out the white balance values recommended by the algorithm, before integrating any functionality into Node-RED or anywhere else - especially since Node-RED is very annoying to develop collaboratively between multiple users on Git.

chevreuill3000, 2026-11-12:
The calibration of the red and the blue gains are done in one click.

closed 2026-11-12 by chevreuill3000