fairscope/PlanktoScope #210: v2023.9.0-beta.0 Node-RED dashboard fails to determine camera type and pixel size

ethanjli, 2023-09-06:
**Describe the bug**
In v2023.9.0-beta.0, the Node-RED dashboard no longer is able to determine the camera type; this causes various display errors in the dashboard, and any raw image datasets acquired with the dashboard are no longer processable by the segmenter (due to a missing metadata field for the camera pixel size calibration).

**To Reproduce**
Steps to reproduce the behavior:
1. Start the PlanktoScope running the adafruit hat version of the software
2. Go to the "Optic Configuration" page. In the "Optic Characterization" panel, the "objective lens aperture" field just says "mm" without a number, the "magnification" field just says "X" without a number, the "pixel resolution" field just says "um" without a number, the "smallest cells to explore" field says "ERROR um", and the "biggest cells to explore" field says "ERROR um"
3. Go to the "Fluidic Acquisition" page. The "total imaged volume" field says "The camera is not known to this system"
4. Go to the "Hardware Settings" page. The hardware version is listed as "Select option", and the "pixel size calibration" field is empty.
5. Start an image acquisition with 10 images, and wait for it to complete successfully.
6. Start a segmentation of the dataset you just acquired. The Node-RED dashboard will soon display an `An exception was raised during the segmentation: unsupported operand type(s) for /: 'int' and 'str'.` error.

**Expected behavior**
Valid information should be displayed in all fields of the Node-RED dashboard listed above. The segmenter should not crash.

**Additional context**
Here is the segmenter's log output, with a traceback of the error which occurs:
```
2023-09-05 21:19:47.232 | INFO     | __main__:<module>:49 - Starting the PlanktoScope data processing segmenter!
2023-09-05 21:19:47.233 | INFO     | __main__:<module>:61 - Welcome!
2023-09-05 21:19:47.235 | INFO     | __main__:<module>:62 - Initialising signals handling and sanitizing the directories (step 1/2)
2023-09-05 21:19:47.238 | INFO     | __main__:<module>:77 - Starting the segmenter control process (step 2/2)
2023-09-05 21:19:47.239 | INFO     | planktoscope.segmenter:__init__:94 - planktoscope.segmenter is initialising
2023-09-05 21:19:47.241 | SUCCESS  | planktoscope.segmenter:__init__:134 - planktoscope.segmenter is initialised and ready to go!
2023-09-05 21:19:47.255 | SUCCESS  | __main__:<module>:86 - Looks like the segmenter is set up and running, have fun!
2023-09-05 21:19:47.264 | INFO     | planktoscope.segmenter:run:1006 - The segmenter control thread has been started in process 28017
2023-09-05 21:19:47.269 | INFO     | planktoscope.mqtt:connect:105 - trying to connect to 127.0.0.1:1883
2023-09-05 21:19:47.286 | SUCCESS  | planktoscope.mqtt:on_connect:131 - segmenter_client connected to 127.0.0.1:1883! - 0: Connection successful
2023-09-05 21:19:47.287 | INFO     | planktoscope.segmenter:run:1018 - Setting up the streaming server thread
2023-09-05 21:19:47.304 | SUCCESS  | planktoscope.segmenter:run:1039 - Segmenter is READY!
2023-09-05 21:19:47.336 | SUCCESS  | planktoscope.mqtt:on_subscribe:142 - segmenter_client subscribed to segmenter/#! - mid:2 qos:(0,)
2023-09-05 21:20:32.587 | INFO     | planktoscope.mqtt:on_message:150 - segmenter_client: segmenter/segment 0 b'{"action":"segment","path":["/home/pi/data/img/2023-09-05"],"settings":{"force":false,"recursive":true,"ecotaxa":true,"keep":true,"process_id":1}}'
2023-09-05 21:20:32.589 | DEBUG    | planktoscope.mqtt:on_message:154 - command is segment
2023-09-05 21:20:32.592 | DEBUG    | planktoscope.mqtt:on_message:157 - args are {'action': 'segment', 'path': ['/home/pi/data/img/2023-09-05'], 'settings': {'force': False, 'recursive': True, 'ecotaxa': True, 'keep': True, 'process_id': 1}}
2023-09-05 21:20:32.594 | DEBUG    | planktoscope.mqtt:on_message:159 - msg is {'topic': 'segmenter/segment', 'payload': {'action': 'segment', 'path': ['/home/pi/data/img/2023-09-05'], 'settings': {'force': False, 'recursive': True, 'ecotaxa': True, 'keep': True, 'process_id': 1}}} or <paho.mqtt.client.MQTTMessage object at 0xe67acdf0>
2023-09-05 21:20:32.869 | INFO     | planktoscope.segmenter:treat_message:923 - We received a new message
2023-09-05 21:20:32.871 | DEBUG    | planktoscope.segmenter:treat_message:925 - {'action': 'segment', 'path': ['/home/pi/data/img/2023-09-05'], 'settings': {'force': False, 'recursive': True, 'ecotaxa': True, 'keep': True, 'process_id': 1}}
2023-09-05 21:20:32.874 | DEBUG    | planktoscope.mqtt:read_message:177 - clearing the __new_message flag
2023-09-05 21:20:32.879 | INFO     | planktoscope.segmenter:segment_list:766 - The pipeline will be run in 3 directories
2023-09-05 21:20:32.881 | DEBUG    | planktoscope.segmenter:segment_list:767 - Those are ['/home/pi/data/img/2023-09-05', "/home/pi/data/img/2023-09-05/Project's_name_1", "/home/pi/data/img/2023-09-05/Project's_name_1/Project's_name_1_1"]
2023-09-05 21:20:32.885 | INFO     | planktoscope.segmenter:segment_list:774 - The process_uuid of this run is test-machine
2023-09-05 21:20:32.889 | INFO     | planktoscope.segmenter:segment_list:775 - The process_id of this run is 1
2023-09-05 21:20:32.892 | DEBUG    | planktoscope.segmenter:segment_list:779 - /home/pi/data/img/2023-09-05: Checking for the presence of metadata.json
2023-09-05 21:20:32.894 | DEBUG    | planktoscope.segmenter:segment_list:798 - Moving to the next folder, /home/pi/data/img/2023-09-05 has no metadata.json
2023-09-05 21:20:32.896 | DEBUG    | planktoscope.segmenter:segment_list:779 - /home/pi/data/img/2023-09-05/Project's_name_1: Checking for the presence of metadata.json
2023-09-05 21:20:32.898 | DEBUG    | planktoscope.segmenter:segment_list:798 - Moving to the next folder, /home/pi/data/img/2023-09-05/Project's_name_1 has no metadata.json
2023-09-05 21:20:32.899 | DEBUG    | planktoscope.segmenter:segment_list:779 - /home/pi/data/img/2023-09-05/Project's_name_1/Project's_name_1_1: Checking for the presence of metadata.json
2023-09-05 21:20:32.901 | DEBUG    | planktoscope.segmenter:segment_list:783 - /home/pi/data/img/2023-09-05/Project's_name_1/Project's_name_1_1: Checking for the presence of done.txt or forcing(False)
2023-09-05 21:20:32.910 | INFO     | planktoscope.segmenter:segment_path:818 - Loading the metadata file for /home/pi/data/img/2023-09-05/Project's_name_1/Project's_name_1_1
2023-09-05 21:20:32.914 | DEBUG    | planktoscope.segmenter:segment_path:821 - Configuration loaded is {'description': {'sample_project': "Project's name", 'sample_id': 'Sample ID', 'sample_uuid': 'Sample UUID (Autogenerated)', 'sample_ship': "Ship's name", 'sample_operator': "Operator's name", 'sample_sampling_gear': 'Sampling gear used', 'sample_concentrated_sample_volume': 'Volume of concentrated sample, in mL', 'sample_total_volume': 'Total volume filtered by the net used, in L', 'sample_dilution_factor': 'Dilution factor of the sample, 0.5 if diluted by 2, 2 if concentrated by 2', 'sample_speed_through_water': 'Speed of the boat through water when sampling, in kts', 'acq_id': 'Acquisition ID', 'acq_uuid': 'Acquisition UUID (Autogenerated)', 'acq_instrument': 'Instrument type', 'acq_instrument_id': 'Instrument ID', 'acq_celltype': 'Flow cell dimension thickness, in µm', 'acq_minimum_mesh': 'Minimum filtration mesh size, in µm', 'acq_maximum_mesh': 'Maximum filtration mesh size, in µm', 'acq_min_esd': '', 'acq_max_esd': '', 'acq_volume': 'Pumped volume, in mL', 'acq_imaged_volume': 'Total imaged volume, in mL', 'acq_magnification': 'Optical magnification', 'acq_fnumber_objective': 'Focal length of the objective, in mm', 'acq_camera_name': 'Name of the camera used', 'acq_nb_frame': 'Number of picture taken', 'acq_local_datetime': 'Instrument local datetime', 'acq_camera_resolution': 'Resolution of the images', 'acq_camera_iso': 'ISO Number of the images', 'acq_camera_shutter_speed': 'Shutter speed of the images, in µs', 'acq_software': 'Software version number', 'object_date': 'Sample collection date (or beginning if using a net)', 'object_time': 'Sample collection time (or beginning if using a net)', 'object_lat': 'Sample collection latitude (or beginning if using a net)', 'object_lon': 'Sample collection longitude (or beginning if using a net)', 'object_depth_min': 'Sample collection minimal depth, in m', 'object_depth_max': 'Sample collection maximum depth, in m', 'process_pixel': 'Pixel imaging resolution, in µm/pixel', 'process_datetime': 'Segmentation timestamp', 'process_id': 'Segmentation ID', 'proc [... cut ...]
                       └ <SegmenterProcess name='segmenter' parent=27904 started>

  File "/home/pi/device-backend/processing/segmenter/planktoscope/segmenter/__init__.py", line 482, in _slice_image
    minsizepix = (minESD / pixel_size) ** 2
                  │        └ 'ERROR'
                  └ 20

TypeError: unsupported operand type(s) for /: 'int' and 'str'
2023-09-05 21:20:52.516 | ERROR    | planktoscope.segmenter:segment_list:795 - There was an error while segmenting /home/pi/data/img/2023-09-05/Project's_name_1/Project's_name_1_1
```

fabienlombard, 2023-09-07:
stange on the Pscope hat it seems to be not totally the same error (got possibility to impose the objective and pixel size, recognize the resolution of the camera , but not its type) and the segmenter is able to process

ethanjli, 2023-09-08:
Based on your observation, I think it's very possible that the problem with determining the pixel size on certain hardware versions is just that the default hardware settings files for v2.1 and v2.3 (which I had copied moved from this GitHub repository into the new repository for the Python backend) are missing a value for the default pixel size: https://github.com/PlanktoScope/device-backend/tree/main/default-configs

ethanjli, 2023-09-12:
Regarding camera detection: the Node-RED dashboard receives information about the camera from the Python hardware controller. Specifically, once the imager process of the hardware controller begins, it will send a MQTT message exactly once at startup with the camera name, before going into its event loop.

Because I have configured systemd to always start the Node-RED dashboard after the backend, the dashboard is probably always  missing that MQTT message. A workaround would be to change the systemd service configuration so that the backend always starts after Node-RED, but that may not necessarily be robust if Node-RED isn't ready to receive MQTT messages by the time the backend starts sending them.

Ultimately, a proper fix to this issue would be to allow the Node-RED dashboard to query the state of the hardware controller. But the MQTT API is not well-suited to this, though it is possible. Ideally we'd have a REST API for that kind of thing, but we may still need some intermediate solution with our MQTT API.

PR #223 (2023-09-14): Fix some Node-RED usability papercuts with a limited scope of changes
This PR:
- Changes the default flowcell type for pscopehat configs to 300 um capillary (fixes #208)
- Adds a description to the "Administration" page describing all the new links we have to the various places where various logs can now be accessed (hopefully fixes #222)
- Explicitly  mentions that existing white balance settings will be overwritten when a hardware version is selected on the "Hardware Settings" page, along with all other hardware settings (fixes #209)
- Bumps device-backend to a version which includes a pixel [... cut]
closed 2023-09-14 by ethanjli