fairscope/PlanktoScope #220: Node-RED dashboard lies to the user about the state of the LED after restarting the Python backend

ethanjli, 2023-09-07:
**Describe the bug**
After the Python backend (in v2.3 of the software and earlier versions) or the Python hardware controller (after v2.3 of the software) is restarted, it turns the sample illumination LED off. But if the sample illumination LED had previously been turned on in the Node-RED dashboard, the Node-RED dashboard still reports to the user that the sample illumination LED is still on - but if you actually look at the LED, you can see that it's off. So the dashboard is now lying to the user about the state of the LED. Anyways, the camera preview screen goes black (as one would expect if they saw what was happening), because there is no light going into the camera after the sample illumination LED has been turned off. The only way to turn the LED back on is to toggle the LED again in the Node-RED dashboard, by telling Node-RED to turn the LED off and then telling Node-RED to turn the LED back on.

I have seen this issue a lot in v2.3 of the software, especially with the Adafruit HAT version of the software (but, if I remember correctly, also with the pscope custom HAT version of the software). Ana got confused by this problem while testing v2023.9.0-beta.0 of the software, [reporting](https://planktoscope.slack.com/archives/C01V5ENKG0M/p1694091490308379?thread_ts=1693848326.637039&cid=C01V5ENKG0M) in the `#6-dev-software` channel on the PlanktoScope Slack: "Then tried the “Restart Hardware” option, new to me, and the Optics preview was completely unresponsive when I went back tot it. As it can be seen below, the light is on but the preview window is black."

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the PlanktoScope.
2. Go to the "Optic Configuration" page. Confirm that the camera preview is a black rectangle.
3. Press the Node-RED toggle to turn the LED on. Confirm that the LED turns on in real life. Confirm that visual content shows up in the camera preview (it's not just a black rectangle anymore).
4. Go to the "Administration" page. Press the "Restart Hardware Controller" button. Confirm that the sample illumination LED turns off in real life.
5. Go back to the "Optic Configuration" page. Confirm that the camera preview is now just a black rectangle again. Confirm that the Node-RED dashboard says that the sample illumination LED is still on.
6. Press the Node-RED toggle to turn the LED off. Confirm that the camera preview is still just a black rectangle, and that the LED is still off in real life.
7. Press the Node-RED toggle to turn the LED on again. Confirm that the LED turns on in real life. Confirm that visual content shows up in the camera preview (it's not just a black rectangle anymore).

**Expected behavior**
The Node-RED dashboard should accurately report to the user about when the sample illumination LED has been turned off by the Python hardware controller.

We will also need to decide whether starting/restarting the Python hardware controller should turn off the sample illumination LED (which is the current behavior), or whether it should keep the previous state of the sample illumination LED.

ethanjli, 2024-04-04:
@Oumayma-hy Here's an example of an issue which requires changes to the MQTT API so that we can fix this issue

ethanjli, 2024-04-04:
@melissadjadoun Here's an example of an issue which requires changes to the MQTT API so that we can fix this issue

chevreuill3000, 2026-11-09:
Closing as a duplicate of #933 (restore the hardware state in the dashboard). See also #1066 for the LED status at startup.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-11-09 by chevreuill3000

chevreuill3000, 2026-11-09:
Closing as a duplicate of #933 (restore the hardware state in the dashboard). See also #1066 for the LED status at startup.